rscada/libmbus #80: Carlo Gavazzi Mbus protocol

ado464, 2015-03-06:
Hello i have a bunch of documentation on VIF's for Carlo Gavazzi meters EM21/EM24/EM33 and i wonder if it is possible to implement this into the coding?

http://www.support-carlogavazzi.se/downloads/Controls/Protokoll%20VMU-B01%20M-BusProtocol.pdf

Cheers,
Ado

lategoodbye, 2015-03-07:
Hi Ado,

currently i don't have the time and resources to implement for such a feature. But here are some ideas:

1. Try to capture some data frames and test them with mbus_parse_hex
2. Extend the vif_table in mbus_protocol_aux.c according to the document

Btw: i already implemented the product strings for these meters in my repo.

Regards
Stefan

rscada, 2015-03-07:
Hi Ado

Unfortunately I do also not have time to work on libmbus right now, but if you are interested in working on this PRs would be welcome. I'm far behind reviewing PR requests too, and I think @lategoodbye fork is more up-to-date.

@lategoodbye, I've given push permissions to this repository to you, feel free to push your updates into this repository and any other changes (current and future), if you are interested.

lategoodbye, 2015-03-07:
@rscada Thank you

ado464, 2015-03-10:
Oh okay guys, so what would be the easiest way on implementing this? All that would be really important are the current W,kWh,Volt, Amp. Can i just add the hex coding beneath this line "mbus_variable_vif vif_table[] = {"? Cheers and thanks

ado464, 2015-03-11:
Like how would these be written in?

Should it be added with the rest of the "W" like this:

    /* E010 1nnn    Power W (0.001W to 10000W) */
    { 0x28, 1.0e-3, "W", "Power" },
    { 0x29, 1.0e-2, "W", "Power" },
    { 0x2A, 1.0e-1, "W", "Power" },
    { 0x2B, 1.0e0,  "W", "Power" },
    { 0x2C, 1.0e1,  "W", "Power" },
    { 0x2D, 1.0e2,  "W", "Power" },
    { 0x2E, 1.0e3,  "W", "Power" },
    { 0x2F, 1.0e4,  "W", "Power" },
    { 0x2Ah, 1.0e0, "W", "Power" },

Or in some other way?

Cheers

lategoodbye, 2015-03-12:
Hi Ado,

W,kWh,Volt, Amp belong to M-Bus standard so they are already implemented in vif_table. It's a list ordered by VIF and separated in 3 areas ( 0x00 - 0xFF Primary VIF, 0x100 - 0x1FF VIFE = FD, 0x200 - 0x2FF VIFE = FB).

BR Stefan