rscada/libmbus #81: Kamstrup Multical 302

ado464, 2015-04-02:
Hello all, i have a Kamstrup multical 302 and i have trouble reading from it or even locating it in my network, but all other 3 Kamstrup Multical 602 are located without trouble, could someone please help me out real quick what it could be?

I have tried swapping the connections on the m-bus as well, i know they are polarity independent but it was worth a try.

best regards,
Ado

lategoodbye, 2015-04-02:
Hi Ado,

could you be please be more specific.

Which program?
Which version?
What is the output?
Have you tried verbose mode?
Which baudrate?

Stefan

ado464, 2015-04-02:
I am running the libmbus 0.8.0 with mbus-serial-scan -d /dev/ttyUSB0 command, it is supposed to be located at the address 183 but it goes past it like it isn't even there, standard baud so i guess it's on 9600.

How do i do it in verbose?

regards,
Ado

lategoodbye, 2015-04-02:
Okay, you already enable debug ("verbose mode") with parameter d.

Here are some tips:
- use current libmbus version from github (0.8.0 is really old)
- set retry to 1 for the primary scan
- try typical baudrate 2400 baud and other
- try secondary scan with mbus-serial-scan-secondary
- disconnect all other M-Bus devices and use a broadcast request

ado464, 2015-04-02:
Updated the libmbus and ran it with the baudrate of 2400 and the meter responded, i guess it can only handle 2400. Thanks for all you're help Stefan! Amazing and quick :D

closed 2015-04-02 by ado464

lategoodbye, 2015-04-02:
Hi Ado,

it would be nice if you could capture the response (e.g. mbus-serial-request-data-multi-reply) of a Kamstrup Multical 302 and post it here as a hex dump.

Stefan

ado464, 2015-04-02:
No problem here is the output:

<MBusData>

    <SlaveInformation>
        <Id>67211183</Id>
        <Manufacturer>KAM</Manufacturer>
        <Version>48</Version>
        <ProductName></ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>3</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Fabrication number</Unit>
        <Value>67211183</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>3219</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>17741</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>14938</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (1e-2  m^3)</Unit>
        <Value>32766</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>On time (hours)</Unit>
        <Value>384</Value>
        <Timestamp>2015-04-02T13:10:55</Timestamp>
    </DataRecord>
[... 161 lines cut ...]
	    <DataRecord id="27">
	        <Function>Instantaneous value</Function>
	        <StorageNumber>0</StorageNumber>
	        <Unit>Manufacturer specific</Unit>
	        <Value>13302000</Value>
	        <Timestamp>2015-04-02T13:10:55</Timestamp>
	    </DataRecord>

	    <DataRecord id="28">
	        <Function>Instantaneous value</Function>
	        <StorageNumber>0</StorageNumber>
	        <Unit>Manufacturer specific</Unit>
	        <Value>6145</Value>
	        <Timestamp>2015-04-02T13:10:55</Timestamp>
	    </DataRecord>

	    <DataRecord id="29">
	        <Function>Instantaneous value</Function>
	        <StorageNumber>0</StorageNumber>
	        <Unit>Firmware version</Unit>
	        <Value>1793</Value>
	        <Timestamp>2015-04-02T13:10:55</Timestamp>
	    </DataRecord>

	</MBusData>

lategoodbye, 2015-04-02:
Hi Ado,

sorry i mean the raw M-Bus telegram (meter response) as hex dump.

Somethings like this:

https://github.com/rscada/libmbus/blob/master/test/test-frames/kamstrup_multical_601.hex

Stefan

ado464, 2015-04-02:
How do i get just that from the device?

regards,
Ado

lategoodbye, 2015-04-02:
The debug parameter should do that:

mbus-serial-request-data-multi-reply -d -b 2400 /dev/ttyUSB0 mbus-address

ado464, 2015-04-02:
There we go, hope it's this:

init_slaves: debug: sending init frame #1
[2015-04-02 22:48:59] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2015-04-02 22:48:59] SEND (005): 10 40 FD 3D 16
[2015-04-02 22:48:59] SEND (005): 10 7B B7 32 16
[2015-04-02 22:49:00] RECV (205): 68 C7 C7 68 08 B7 72 83 11 21 67 2D 2C 30 04 21 00 00 00 0C 78 83 11 21 67 04 06 40 0D 00 00 04 86 FF 02 00 00 00 00 04 FF 07 9A 49 00 00 04 FF 08 10 3E 00 00 04 14 FC 87 00 00 04 22 89 01 00 00 34 22 02 00 00 00 02 59 1A 15 02 5D 40 12 02 61 DA 02 04 2D B4 00 00 00 14 2D 1D 01 00 00 04 3B 67 08 00 00 14 3B 23 09 00 00 02 FD 17 00 00 04 6D 30 37 E2 14 44 06 A9 09 00 00 44 86 FF 02 00 00 00 00 44 FF 07 E7 33 00 00 44 FF 08 7E 2B 00 00 44 14 FE 5F 00 00 54 2D 26 01 00 00 54 3B 17 09 00 00 42 6C FF 13 07 FF 0F AF 8F 01 04 00 00 00 00 04 FF 11 B2 AC 00 00 04 FF 12 F0 F8 CA 00 02 FF 1A 01 18 02 FD 0E 01 07 F6 16
mbus_frame_print: Dumping M-Bus frame [type 4, 205 bytes]: 68 C7 C7 68 08 B7 72 83 11 21 67 2D 2C 30 04 21 00 00 00 0C 78 83 11 21 67 04 06 40 0D 00 00 04 86 FF 02 00 00 00 00 04 FF 07 9A 49 00 00 04 FF 08 10 3E 00 00 04 14 FC 87 00 00 04 22 89 01 00 00 34 22 02 00 00 00 02 59 1A 15 02 5D 40 12 02 61 DA 02 04 2D B4 00 00 00 14 2D 1D 01 00 00 04 3B 67 08 00 00 14 3B 23 09 00 00 02 FD 17 00 00 04 6D 30 37 E2 14 44 06 A9 09 00 00 44 86 FF 02 00 00 00 00 44 FF 07 E7 33 00 00 44 FF 08 7E 2B 00 00 44 14 FE 5F 00 00 54 2D 26 01 00 00 54 3B 17 09 00 00 42 6C FF 13 07 FF 0F AF 8F 01 04 00 00 00 00 04 FF 11 B2 AC 00 00 04 FF 12 F0 F8 CA 00 02 FF 1A 01 18 02 FD 0E 01 07 F6 16 

regards,
Ado

lategoodbye, 2015-04-03:
Thank you very much Ado