rscada/libmbus #83: Multiple VIFE Handling

Chucky89, 2015-06-18:
Hello.

My Issue is more of a question I suppose. I have a special meter whit lots of costum VIFE's. When I run the normal scan application most units are printed out as manufacturer specific.

What would be the right way to patch this in? My first try to add them to the function "mbus_vib_unit_lookup" in mbus-protocol.c resulted in printing only the last VIFE, when there are more then one in the telegram.

Greetings

lategoodbye, 2015-06-19:
Hi  Chucky,

first of all you will need to identify your special meter. Look at mbus_data_product_name() for examples.

Personally i recommend to use mbus-protocol-aux.c for new features. It has table based VIF handling. But i don't know if it fits to your needs.

Finally i want to mention that i can't guarantee that patches for specific meters are accepted for libmbus.

Regards
Stefan

Chucky89, 2015-06-24:
It's a "Gossen Metrawatt" meter, model U189.

My current goal is to optimize the XML output so one can add every possible manufacturer specific unit output to the source files as a patch. Currently trying to wrap my head around that table based handling and how I could filter by manufacturer name in the easiest and most understandable way.

Btw: I get the "Failed to receive M-Bus response frame." error everytime I request data for the first time.. Contacting the meter with "MBCONF" first solves this, don't exactly know why.

lategoodbye, 2015-06-25:
Hi Chucky,

according to the error: do you use the current version from Github?
Could you please trace the frame ( usual add parameter -v ) and open a separate ticket?

Regards
Stefan