rscada/libmbus #84: Failed to recieve M-Bus response frame

Chucky89, 2015-06-26:
Hello.

Every time I try to connect to the Meter ("Gossen Metrawatt") for the first time after turning on, I get the following error, using the function ```mbus_read_slave```:

```c
[2015-06-26 08:32:54] SEND (005): 10 40 FD 3D 16
[2015-06-26 08:32:55] SEND (005): 10 40 FD 3D 16
[2015-06-26 08:32:55] SEND (005): 10 5B 04 5F 16
mbus_read_slave: Failed to receive M-Bus response frame.
init_slaves: debug: sending init frame #1
init_slaves: debug: sending init frame #2
M-bus data parse error: Got null pointer to frame, data or zero data_size.
```
I can only recieve the frame if I first run a query via the Windows configuration tool MBCONF. Afterwards everything runs as expected.
```c
[2015-06-26 08:43:22] SEND (005): 10 40 FD 3D 16
[2015-06-26 08:43:22] SEND (005): 10 40 FD 3D 16
[2015-06-26 08:43:23] SEND (005): 10 5B 04 5F 16
[2015-06-26 08:43:24] RECV (210): 68 CC CC 68 08 04 72 73 16 00 00 A3 1D 02 02 01 20 00 00 01 FF 54 01 02 FF 52 05 00 01
 FD DC FF 64 01 01 FF 61 00
 01 FF 62 00 06 FF 63 00 00 01 00 00 00 86 10 82 FF 80 FF 00 34 A2 01 00 00 00 86 10 82 FF 81 FF 00 B4 71 00 00 00 00 86 20 82 FF 80 FF 00 00 00 00 00 00 00 86 20 82 FF 81 FF 00 00 00 00 00 00 00 86 90 00 FF 93 FF 10 03 4E 00 00 00 00 86 90 00 FF 93 FF 20 67 51 00 00
 00 00 86 A0 00 FF 93 FF 10 00 00 00 00 00 00 86 A0 00 FF 93 FF 20 00 00 00 00 00 00 86 90 00 FF 93 FF 30 45 69 00 00 00 00 86 90 00 FF 93 FF 40 95 20 00 00 00 00 86 A0 00 FF 93 FF 30 00 00 00 00 00 00 86 A0 00 FF 93 FF 40 00 00 00 00 00 00 0F AC 16
init_slaves: debug: sending init frame #1
init_slaves: debug: sending init frame #2
mbus_frame_print: Dumping M-Bus frame [type 4, 210 bytes]: 68 CC CC 68 08 04 72 73 16 00 00 A3 1D 02 02 01 20 00 00 01 FF 54 01 02 FF 52 05 00 01 FD DC FF 64 01 01 FF 61 00 01 FF 62 00 06 FF 63 00 00 01 00 00 00 86 10 82 FF 80 FF 00 34 A2 01 00 00 00 86 10 82 FF 81 FF 00 B4 71 00 00 00 00 86 20 82 FF 80 FF 00 00 00 00 00 00 00 86 20 82 FF 81 FF 00 00 00 00 00 00 00 86 90 00 FF 93 FF 10 03 4E 00 00 00 00 86 90 00 FF 93 FF 20 67 51 00 00 00 00 86 A0 00 FF 93 FF 10 00 00 00 00 00 00 86 A0 00 FF 93 FF 20 00 00 00 00 00 00 86 90 00 FF 93 FF 30 45 69 00 00 00 00 86 90 00 FF 93 FF 40 95 20 00 00 00 00 86 A0 00 FF 93 FF 30 00 00 00 00 00 00 86 A0 00 FF 93 FF 40 00 00 00 00 00 00 0F AC 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>1673</Id>
        <Manufacturer>GMC</Manufacturer>
        <Version>2</Version>
        <ProductName>Firmware Version 2</ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>1</AccessNumber>
        <Status>20</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit></Unit>
        <Quantity>Manufacturer specific</Quantity>
        <Value>1.000000</Value>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit></Unit>
        <Quantity>Manufacturer specific</Quantity>
        <Value>5.000000</Value>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>A</Unit>
        <Quantity>Current</Quantity>
        <Value>1.000000</Value>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit></Unit>
        <Quantity>Manufacturer specific</Quantity>
        <Value>0.000000</Value>
    </DataRecord>

(Many Data records)
....

</MBusData>
```

The libmbus app itself is running on a Wago 750-8202 SPS. Hope you can give me a hand on this error, as I don't know if the problem is the libmbus or the meter itself.

lategoodbye, 2015-06-26:
Hi Chucky,

looks like a problem of libmbus, because MBCONF doesn't fail. Unfortunaley i don't have any M-Bus hardware available. Could you trace the M-Bus messages of MBCONF?

Stefan

Chucky89, 2015-06-26:
Of course, here it is:

```

TX-Buffer:
10 7B 04 7F 16

RX-Buffer:
68 CC CC 68 08 04 72 73 16 00 00 A3 1D 02 02 00 20 00 00 01 FF 54 01 02 FF 52 05 00 01 FD
DC FF 64 01 01 FF 61 00 01 FF 62 00 06 FF 63 00 00 01 00 00 00 86 10 82 FF 80 FF 00 34 A2 01
00 00 00 86 10 82 FF 81 FF 00 B4 71 00 00 00 00 86 20 82 FF 80 FF 00 00 00 00 00 00 00 86 20
82 FF 81 FF 00 00 00 00 00 00 00 86 90 00 FF 93 FF 10 03 4E 00 00 00 00 86 90 00 FF 93 FF 20
67 51 00 00 00 00 86 A0 00 FF 93 FF 10 00 00 00 00 00 00 86 A0 00 FF 93 FF 20 00 00 00 00 00
00 86 90 00 FF 93 FF 30 45 69 00 00 00 00 86 90 00 FF 93 FF 40 95 20 00 00 00 00 86 A0 00 FF
93 FF 30 00 00 00 00 00 00 86 A0 00 FF 93 FF 40 00 00 00 00 00 00 0F AB 16
```

It occurred to me that the MBCONF sends a slightly different request message - 7B and 7F instead of 5B and 5F.

lategoodbye, 2015-06-27:
Looks like a problem with Frame Count Bit handling. I will try to fix it.

lategoodbye, 2015-06-27:
@Chucky89 

Please try branch https://github.com/rscada/libmbus/tree/fix-fcb-handling

The changes are only compile tested.

Chucky89, 2015-07-03:
Sorry to report that the outcome is the same. No response frame recieved.

I'm trying to locate the error parallely.

lategoodbye, 2015-07-03:
Hi Chucky,

maybe it's a timing issue. If you have the equipment, you could try to trace the UART on your M-Bus master.