rscada/libmbus #90: Sontex Supercal531 parse error: Unknown control information 0xb7

arska, 2016-01-31:
$ ./mbus_parse_hex test-frames/SON-Sontex-Supercal531-multiframe.hex 
int mbus_frame_print(mbus_frame *): Dumping M-Bus frame [type 4, 191 bytes]: 68 B9 B9 68 08 09 B7 01 01 04 40 83 03 C0 00 00 69 92 30 13 70 92 30 13 07 0C 1F 21 00 00 00 00 00 00 00 00 6C 05 00 00 04 94 F2 0C 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 0E 02 00 00 00 00 00 00 00 0E 9E 03 00 00 6C 05 00 00 6C 05 00 00 62 05 00 00 5C 05 00 00 ED 04 00 00 6A 04 00 00 9E 03 00 00 7D 03 00 00 7D 03 00 00 7D 03 00 00 7C 03 00 00 7C 03 00 00 7C 03 00 00 7C 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1F 36 16 
mbus_frame_data_parse: Unknown control information 0xb7

sometimes it works, sometimes it doesn't. I'll be working on this, ping me if you know something :)

68 90 90 68 08 0A 72 01 46 63 12 EE 4D 0D 0C 4A 00 00 00 C5 89 01 63 3D 4A B5 3F 85 8A 01 63 C9 B2 B9 3F C5 8A 01 63 98 C5 B7 3F 85 8B 01 63 4F 07 B8 3F C5 8B 01 63 1A C1 B8 3F 85 8C 01 63 FB 97 BA 3F C5 8C 01 63 18 EE BD 3F 85 8D 01 63 55 46 B6 3F C5 8D 01 63 07 DD BE 3F 85 8E 01 63 A0 2F BA 3F C5 8E 01 63 07 FF BB 3F 85 8F 01 63 80 5E BB 3F C5 8F 01 63 66 22 BE 3F 85 80 02 63 F4 37 C4 3F C5 80 02 63 03 DF BC 3F 85 81 02 63 B0 D9 C4 3F 1F 4A 16

68 90 90 68 08 09 72 69 92 30 13 EE 4D 0D 04 DA 00 00 00 C5 89 01 63 00 00 00 00 85 8A 01 63 00 00 00 00 C5 8A 01 63 00 00 00 00 85 8B 01 63 00 00 00 00 C5 8B 01 63 00 00 00 00 85 8C 01 63 00 00 00 00 C5 8C 01 63 00 00 00 00 85 8D 01 63 00 00 00 00 C5 8D 01 63 00 00 00 00 85 8E 01 63 00 00 00 00 C5 8E 01 63 00 00 00 00 85 8F 01 63 00 00 00 00 C5 8F 01 63 00 00 00 00 85 80 02 63 00 00 00 00 C5 80 02 63 00 00 00 00 85 81 02 63 00 00 00 00 1F 39 16

68 90 90 68 08 06 72 81 25 00 13 EE 4D 0D 04 8B 00 00 00 C5 81 02 5F D2 4A 46 42 85 82 02 5F 8F 2E 68 42 C5 82 02 5F C3 6F 5F 42 85 83 02 5F CD F2 62 42 C5 83 02 5F EA 3A 66 42 85 84 02 5F 93 CA 52 42 C5 84 02 5F 79 C1 2E 42 85 85 02 5F 25 C2 52 42 C5 85 02 5F 41 09 11 42 85 86 02 5F 66 0F 01 42 C5 86 02 5F ED D2 0B 42 85 87 02 5F 30 55 2D 42 C5 87 02 5F 50 79 5F 42 85 88 02 5F B5 1E 64 42 C5 88 02 5F 49 00 66 42 85 89 02 5F 7C 50 4F 42 1F 88 16

68 BA BA 68 08 08 B7 01 02 E5 00 D3 00 40 19 59 42 00 69 32 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E2 15 96 46 82 6B B4 46 D2 63 C1 46 22 1B C4 46 E2 48 CC 46 C2 3B D6 46 21 16 30 46 13 14 1D 21 13 07 1D 21 13 07 1C 21 13 07 1B 21 13 07 1A 21 13 07 19 21 13 0D 17 21 1F 1C 16

68 BA BA 68 08 08 B7 01 02 E5 00 D3 00 C0 B5 60 42 40 46 33 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E2 15 96 46 82 6B B4 46 D2 63 C1 46 22 1B C4 46 E2 48 CC 46 C2 3B D6 46 21 16 30 46 13 14 1D 21 13 07 1D 21 13 07 1C 21 13 07 1B 21 13 07 1A 21 13 07 19 21 13 0D 17 21 1F 5D 16

68 BA BA 68 08 08 B7 01 02 E5 00 D3 00 80 94 60 42 80 15 33 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E2 15 96 46 82 6B B4 46 D2 63 C1 46 22 1B C4 46 E2 48 CC 46 C2 3B D6 46 21 16 30 46 13 14 1D 21 13 07 1D 21 13 07 1C 21 13 07 1B 21 13 07 1A 21 13 07 19 21 13 0D 17 21 1F 0B 16

68 BA BA 68 08 08 B7 01 02 E5 00 D3 00 40 EF 61 42 00 08 30 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E2 15 96 46 82 6B B4 46 D2 63 C1 46 22 1B C4 46 E2 48 CC 46 C2 3B D6 46 21 16 30 46 13 14 1D 21 13 07 1D 21 13 07 1C 21 13 07 1B 21 13 07 1A 21 13 07 19 21 13 0D 17 21 1F 97 16

68 BA BA 68 08 08 B7 01 02 E5 00 D3 00 80 7E 61 42 40 16 30 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E2 15 96 46 82 6B B4 46 D2 63 C1 46 22 1B C4 46 E2 48 CC 46 C2 3B D6 46 21 16 30 46 13 14 1D 21 13 07 1D 21 13 07 1C 21 13 07 1B 21 13 07 1A 21 13 07 19 21 13 0D 17 21 1F B4 16

68 BA BA 68 08 08 B7 01 02 E5 00 D3 00 00 0C 61 42 80 3C 30 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E2 15 96 46 82 6B B4 46 D2 63 C1 46 22 1B C4 46 E2 48 CC 46 C2 3B D6 46 21 16 30 46 13 14 1D 21 13 07 1D 21 13 07 1C 21 13 07 1B 21 13 07 1A 21 13 07 19 21 13 0D 17 21 1F 28 16

68 3E 3E 68 08 08 72 43 25 00 13 EE 4D 0D 04 3F 00 00 00 C4 00 06 28 77 00 00 C4 00 13 C2 B1 19 00 D5 82 03 2B E2 15 96 46 C4 82 03 6D 13 14 1D 21 D5 86 04 3E 10 58 69 3F C4 86 04 6D 13 11 1D 21 0F A0 16

68 3E 3E 68 08 08 72 43 25 00 13 EE 4D 0D 04 40 00 00 00 C4 00 06 28 77 00 00 C4 00 13 C2 B1 19 00 D5 82 03 2B E2 15 96 46 C4 82 03 6D 13 14 1D 21 D5 86 04 3E 10 58 69 3F C4 86 04 6D 13 11 1D 21 0F A1 16

68 50 50 68 08 08 72 43 25 00 13 EE 4D 0D 04 49 00 00 00 02 FD 17 00 00 34 75 DD 01 11 00 04 6D 39 0F 1E 21 04 06 3B 92 00 00 04 13 B8 C6 1D 00 05 5B C0 15 68 42 05 5F 00 26 32 42 05 3E 23 0B C2 3D 05 2B E2 A0 B0 44 03 22 7C 57 00 0C 78 43 25 00 13 1F CA 16

68 B9 B9 68 08 09 B7 01 01 04 40 83 03 C0 00 00 69 92 30 13 70 92 30 13 07 0C 1F 21 00 00 00 00 00 00 00 00 6C 05 00 00 04 94 F2 0C 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 0E 02 00 00 00 00 00 00 00 0E 9E 03 00 00 6C 05 00 00 6C 05 00 00 62 05 00 00 5C 05 00 00 ED 04 00 00 6A 04 00 00 9E 03 00 00 7D 03 00 00 7D 03 00 00 7D 03 00 00 7C 03 00 00 7C 03 00 00 7C 03 00 00 7C 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1F 36 16

arska, 2016-01-31:
Hmm, I think I found something: http://www.enbra.cz/images/stories/INTERNET/KE%20STA%C5%BDEN%C3%8D/Dokumentace%20k%20m%C4%9B%C5%99i%C4%8D%C5%AFm%20tepla/M%C4%9B%C5%99i%C4%8De%20tepla%20soupravy/531440/M-Bus%20Frames%20531%20-%20Rev.%2029-03-2010.pdf page 128

Trying to figure out why the device would answer with C=08, CI=B7 instead of with a regular RSP_UD (C=08, CI=72) next

lategoodbye, 2016-01-31:
That's really strange. 5 years ago i tested libmbus with Sontex Supercal 531 and never noticed this behavior. Maybe this new to the meter firmware or a configuration option. If you need it for a project try to contact the Sontex folks.

arska, 2016-01-31:
I came half way in implementing handling for C=08,CI=B7 when i noticed the init_slaves() function was badly broken: it always sent initialization to the address 253, nobody ever answered and the device was not reset...
have a look at https://github.com/arska/libmbus/commit/694fa54b2e804493be322ae60c9a9d01d6f258d9, i don't know how to create the pull request with just that commit ?

lategoodbye, 2016-01-31:
Please explain why init_slaves() should be broken? The last change has been for 2 years.

Can you try https://github.com/rscada/libmbus/tree/fix-fcb-handling ?

arska, 2016-01-31:
It's a logic bug: the "ping" frame is sent to 253 ("all selected slaves") before any slaves are selected using secondary addressing. So no device resets or responds. Usually that is not a problem and I suspect this was added to "open the communication path" before actually selecting a slave (for secondary addressing) and/or requesting data. The Sontex devices on the other hand _need_ to be reset before they can queried or else they start sending their own data format (the pdf linked above does not state how/why the special data format is queried).

Probably that's why nobody noticed the bus reset wasn't working: because nobody used devices that actually needed to be reset first.

lategoodbye, 2016-02-01:
Yes, the intension behind init_slaves() was to deselect all secondary addresses first before requesting data in order to avoid collisions.

So you suggest to send SND_NKE to 255 ?

arska, 2016-02-01:
A-ha, since I only use primary addressing with these meters I didn't have the problem with "de-selecting previously selected slaves", was that the use-case you meant ?

I just got feedback from the vendor, the meters indeed need a reset (SND_NKE, individually to their primary/secondary address) before querying. Querying them is stateful, they will iterate through 108 different response types with "more records in next telegram" with the last 2 being of the C=08,CI=B7 type.

I'll prepare a patch where I change back the "deselecting" (to 253) and include an optional "send reset before query" command line parameter which will be sent to the primary address or 253 if using secondary addressing.

lategoodbye, 2016-02-01:
Yes, think of a secondary address scan which has been interrupted without deselection the chance of multiple responses is high.

Please make one patch for every semantic change and let them base on rscada/master. This is necessary to make changes bisectable.

lategoodbye, 2016-02-06:
Btw please use parameter -n for more reliable M-Bus frame decoding. And yes it should be changed in generate-xml.sh.

arska, 2016-02-08:
pending #95 for fix

closed 2021-09-17 by arska