rscada/libmbus #97: mbus-tcp-request-data / mbus-tcp-request-data-multi-reply

t006, 2016-03-28:
Here I have a ABB CEM 05100 reading a B23 212-100 meter.
mbus-tcp-request-data is working perfectly, however mbus-tcp-request-data-multi-reply failed to receive any data. 
When I login into the web interface of the CEM I can see a lot more frames. Any idea?

`mbus-tcp-request-data-multi-reply -d 192.168.2.43 6021 1
int init_slaves(mbus_handle *): debug: sending init frame #1
[2016-03-28 06:55:01] SEND (005): 10 40 FD 3D 16
int init_slaves(mbus_handle *): debug: sending init frame #2
[2016-03-28 06:55:06] SEND (005): 10 40 FD 3D 16
[2016-03-28 06:55:06] SEND (005): 10 7B 01 7C 16
int mbus_sendrecv_request(mbus_handle *, int, mbus_frame *, int): Failed to receive M-Bus response frame.
Failed to send/receive M-Bus request.`

`mbus-tcp-request-data -d 192.168.2.43 6021 1 [2016-03-28 06:55:19] SEND (005): 10 7B 01 7C 16
[2016-03-28 06:55:20] RECV (149): 68 8F 8F 68 08 01 72 83 52 07 00 42 04 20 02 36 20 00 00 0E 84 00 03 00 00 00 00 00 8E 40 84 00 00 00 00 00 00 00 04 FF A0 15 00 00 00 00 04 FF A1 15 00 00 00 00 04 FF A2 15 00 00 00 00 04 FF A3 15 00 00 00 00 07 FF A6 00 00 00 00 00 00 00 00 00 07 FF A7 00 06 00 00 00 00 00 00 00 07 FF A8 00 00 00 B6 21 1B 00 00 00 07 FF A9 00 B0 0D D9 00 00 00 00 00 0D FD 8E 00 07 30 2E 37 31 2E 31 42 0D FF AA 00 0B 30 30 31 2D 32 31 32 20 33 32 42 1F 26 16
int mbus_frame_print(mbus_frame *): Dumping M-Bus frame [type 4, 149 bytes]: 68 8F 8F 68 08 01 72 83 52 07 00 42 04 20 02 36 20 00 00 0E 84 00 03 00 00 00 00 00 8E 40 84 00 00 00 00 00 00 00 04 FF A0 15 00 00 00 00 04 FF A1 15 00 00 00 00 04 FF A2 15 00 00 00 00 04 FF A3 15 00 00 00 00 07 FF A6 00 00 00 00 00 00 00 00 00 07 FF A7 00 06 00 00 00 00 00 00 00 07 FF A8 00 00 00 B6 21 1B 00 00 00 07 FF A9 00 B0 0D D9 00 00 00 00 00 0D FD 8E 00 07 30 2E 37 31 2E 31 42 0D FF AA 00 0B 30 30 31 2D 32 31 32 20 33 32 42 1F 26 16
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>75283</Id>
        <Manufacturer>ABB</Manufacturer>
        <Version>32</Version>
        <ProductName></ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>54</AccessNumber>
        <Status>20</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (10 Wh)</Unit>
        <Value>3</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>1</Device>
        <Unit>Energy (10 Wh)</Unit>
        <Value>0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>6</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>116529692672</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>14224816</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Firmware version</Unit>
        <Value>B1.17.0</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>B23 212-100</Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

    <DataRecord id="12">
        <Function>More records follow</Function>
        <Value></Value>
        <Timestamp>2016-03-28T06:55:20</Timestamp>
    </DataRecord>

</MBusData>`

lategoodbye, 2016-03-28:
Do you have the same behavior on requesting the secondary address of the B23 212-100?

t006, 2016-03-28:
Yes, same story

t006, 2016-03-29:
The multi-reply fails on Debian@armv7, Freebsd@intel. However it succeeds on Debian@intel/virtualbox. On all tested platforms mbus-tcp-request-data succeeds

lategoodbye, 2016-03-29:
@t006 Could you please try the following patch?

https://gist.github.com/lategoodbye/79f8b487d78522cdaaeb0de8c857d2a7

arska, 2016-03-29:
Although on page 210 the manual at https://library.e.abb.com/public/94eba55dce8048e6b6e6fcff264a3163/2CMC484001M0201.pdf mentions the meter can be stateful in the M-Bus answers and should be properly reset (where #95 would come in) I don't think this applies here as according to the debug output both applications send out "10 7B 01 7C 16" as the last command and the meter answers to one but not the other. Lets see if the timing helps or if the request just doesn't make it all the way to the reader and needs to be repeated.
@t006 what do you mean by "When I login into the web interface of the CEM I can see a lot more frames" ? Screenshot ?

Cheers,
Aarno