rscada/libmbus #98: Reading Endress Hauser RMS621 M-Bus

ado464, 2016-03-31:
Hello guys, i am having issues reading this meter. Check the output

Reading 1:
<pre>
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>28336</Id>
        <Manufacturer>EAH</Manufacturer>
        <Version>2</Version>
        <ProductName></ProductName>
        <Medium>Bus/System</Medium>
        <AccessNumber>15</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2016-03-31T17:02:00</Value>
        <Timestamp>2016-03-31T15:03:21</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow ( m^3/h)</Unit>
        <Value>166659104.000000</Value>
        <Timestamp>2016-03-31T15:03:21</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (deg C)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2016-03-31T15:03:21</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (deg C)</Unit>
        <Value>-27733311640882380800.000000</Value>
        <Timestamp>2016-03-31T15:03:21</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>-0.000000</Value>
        <Timestamp>2016-03-31T15:03:21</Timestamp>
    </DataRecord>

</MBusData>
</pre>

Reading2:

<pre>
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>28336</Id>
        <Manufacturer>EAH</Manufacturer>
        <Version>2</Version>
        <ProductName></ProductName>
        <Medium>Bus/System</Medium>
        <AccessNumber>3</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2016-03-31T17:06:00</Value>
        <Timestamp>2016-03-31T15:07:21</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow ( m^3/h)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2016-03-31T15:07:21</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (deg C)</Unit>
        <Value>-0.000000</Value>
        <Timestamp>2016-03-31T15:07:21</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (deg C)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2016-03-31T15:07:21</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>-19056912498688.000000</Value>
        <Timestamp>2016-03-31T15:07:21</Timestamp>
    </DataRecord>

</MBusData>
</pre>

The ID=4 isn't even kWh, it's Mw but shows kWh? Also.. negative value?

Also when i scan for the meter it keeps changing meter place? Very weird. It is set to adress 1.

<pre>
Scanning primary addresses:
0 [2016-03-31 15:07:52] SEND (005): 10 40 00 40 16
1 [2016-03-31 15:07:52] SEND (005): 10 40 01 41 16
[2016-03-31 15:07:52] RECV (001): E5
Found a M-Bus device at address 1
2 [2016-03-31 15:07:52] SEND (005): 10 40 02 42 16
3 [2016-03-31 15:07:52] SEND (005): 10 40 03 43 16
4 [2016-03-31 15:07:52] SEND (005): 10 40 04 44 16
5 [2016-03-31 15:07:53] SEND (005): 10 40 05 45 16
6 [2016-03-31 15:07:53] SEND (005): 10 40 06 46 16
7 [2016-03-31 15:07:53] SEND (005): 10 40 07 47 16
8 [2016-03-31 15:07:53] SEND (005): 10 40 08 48 16
9 [2016-03-31 15:07:54] SEND (005): 10 40 09 49 16
10 [2016-03-31 15:07:54] SEND (005): 10 40 0A 4A 16
</pre>

But comes also up as:

<pre>
Scanning primary addresses:
0 [2016-03-31 15:08:21] SEND (005): 10 40 00 40 16
1 [2016-03-31 15:08:21] SEND (005): 10 40 01 41 16
2 [2016-03-31 15:08:22] SEND (005): 10 40 02 42 16
3 [2016-03-31 15:08:22] SEND (005): 10 40 03 43 16
[2016-03-31 15:08:22] RECV (001): E5
Found a M-Bus device at address 3
4 [2016-03-31 15:08:22] SEND (005): 10 40 04 44 16
5 [2016-03-31 15:08:22] SEND (005): 10 40 05 45 16
6 [2016-03-31 15:08:23] SEND (005): 10 40 06 46 16
7 [2016-03-31 15:08:23] SEND (005): 10 40 07 47 16
8 [2016-03-31 15:08:23] SEND (005): 10 40 08 48 16
</pre>

Debug output:

<pre>
mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 1
init_slaves: debug: sending init frame #1
[2016-03-31 15:14:21] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2016-03-31 15:14:22] SEND (005): 10 40 FD 3D 16
[2016-03-31 15:14:22] SEND (005): 10 5B 01 5C 16
[2016-03-31 15:14:22] RECV (051): 68 2D 2D 68 08 01 72 36 83 02 00 28 14 02 0E 01 00 00 00 04 6D 0D 91 1F 23 05 3E 12 3A EB 42 05 5B 1E 8E 7F 42 05 5F FE A6 70 42 05 06 4C 2F B0 44 91 16
mbus_frame_print: Dumping M-Bus frame [type 4, 51 bytes]: 68 2D 2D 68 08 01 72 36 83 02 00 28 14 02 0E 01 00 00 00 04 6D 0D 91 1F 23 05 3E 12 3A EB 42 05 5B 1E 8E 7F 42 05 5F FE A6 70 42 05 06 4C 2F B0 44 91 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>28336</Id>
        <Manufacturer>EAH</Manufacturer>
        <Version>2</Version>
        <ProductName></ProductName>
        <Medium>Bus/System</Medium>
        <AccessNumber>1</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2016-03-31T17:13:00</Value>
        <Timestamp>2016-03-31T15:14:22</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow ( m^3/h)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2016-03-31T15:14:22</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (deg C)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2016-03-31T15:14:22</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (deg C)</Unit>
        <Value>-110617361593742888303337684056997363712.000000</Value>
        <Timestamp>2016-03-31T15:14:22</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>46055696.000000</Value>
        <Timestamp>2016-03-31T15:14:22</Timestamp>
    </DataRecord>

</MBusData>
</pre>

Thanks, much help is appreciated.

Regards,
Ado

lategoodbye, 2016-03-31:
Hi Ado,

could you please try the following patch for the decoding issue:
https://gist.github.com/lategoodbye/5782f4003a13cee0fb5320390372b168

According to the scan issue i can't see any wrong behavior of libmbus.
Have you tried secondary address or another M-Bus tool?

Regards
Stefan

ado464, 2016-03-31:
Thanks for the quick reply Stefan, is that line of code submitted so i can update my git version or do i make the modification myself?

I will try it tonight and report asap. 

Yeah i am suspecting the meter switching the address atound :p

Regards,
Ado

lategoodbye, 2016-03-31:
No, this patch is not submitted as long as i'm unsure about it, but i tested it against the hexdump and the values looks more realistic. So please apply the patch manually and in good case i will submit it.

ado464, 2016-03-31:
Okay i have applied the patch and stuff suddenly became readable ;) looks good, check it out:

<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>28336</Id>
        <Manufacturer>EAH</Manufacturer>
        <Version>2</Version>
        <ProductName></ProductName>
        <Medium>Bus/System</Medium>
        <AccessNumber>5</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2016-03-31T23:02:00</Value>
        <Timestamp>2016-03-31T21:02:55</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow ( m^3/h)</Unit>
        <Value>118.155693</Value>
        <Timestamp>2016-03-31T21:02:55</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (deg C)</Unit>
        <Value>64.074036</Value>
        <Timestamp>2016-03-31T21:02:55</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (deg C)</Unit>
        <Value>59.859058</Value>
        <Timestamp>2016-03-31T21:02:55</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>5018.509766</Value>
        <Timestamp>2016-03-31T21:02:55</Timestamp>
    </DataRecord>

</MBusData>

4th id is still kWh instead of Mw, don't really know what's up with that, will look into it.

Debug:

init_slaves: debug: sending init frame #1
[2016-03-31 21:06:06] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2016-03-31 21:06:06] SEND (005): 10 40 FD 3D 16
[2016-03-31 21:06:06] SEND (005): 10 5B 01 5C 16
[2016-03-31 21:06:06] RECV (051): 68 2D 2D 68 08 01 72 36 83 02 00 28 14 02 0E 06 00 00 00 04 6D 05 97 1F 23 05 3E 26 FD EE 42 05 5B 55 66 7F 42 05 5F 7F 80 6D 42 05 06 47 CE 9D 45 5D 16
mbus_frame_print: Dumping M-Bus frame [type 4, 51 bytes]: 68 2D 2D 68 08 01 72 36 83 02 00 28 14 02 0E 06 00 00 00 04 6D 05 97 1F 23 05 3E 26 FD EE 42 05 5B 55 66 7F 42 05 5F 7F 80 6D 42 05 06 47 CE 9D 45 5D 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>28336</Id>
        <Manufacturer>EAH</Manufacturer>
        <Version>2</Version>
        <ProductName></ProductName>
        <Medium>Bus/System</Medium>
        <AccessNumber>6</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2016-03-31T23:05:00</Value>
        <Timestamp>2016-03-31T21:06:06</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow ( m^3/h)</Unit>
        <Value>119.494431</Value>
        <Timestamp>2016-03-31T21:06:06</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (deg C)</Unit>
        <Value>63.849934</Value>
        <Timestamp>2016-03-31T21:06:06</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (deg C)</Unit>
        <Value>59.375484</Value>
        <Timestamp>2016-03-31T21:06:06</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>5049.784668</Value>
        <Timestamp>2016-03-31T21:06:06</Timestamp>
    </DataRecord>

</MBusData>

Also the infinite numbers after the decimal/dot, is that a meter issue or libmbus, what do you think Stefan? It's not really a issue just interesting, i can probably make it show less with rounding up the numbers :)

Thanks again for the quick response and patch!

Regards,
Ado

lategoodbye, 2016-04-01:
I will take a deeper look at the unit issue.

But i've another question do you have any problems with adding one of your dumps to the test frames?
This might be helpful in order to avoid regressions.

ado464, 2016-04-01:
Thanks again :)

No, no problem at all, what do you need from me and how do i get it for you? :)

Regards,
Ado

ado464, 2016-04-01:
Hello, i have resolved the "problem" with the unit Energy (kWh), it was all in the documentation of the m-bus for Endress Hauser.

The process values are always transmitted in the following basic units:
Volume flow.....m3/h
Temperature.....°C
Pressure.....bar
Heat flow (performance) for steam applications.....MW
Heat flow (performance) for all other applications.....kW
Mass flow for steam applications.....t/h
Mass flow for all other applications.....kg/h
Standard volume.....(N)m3/h
Total volume.....m3
Total mass for steam applications.....t
Total mass for all other applications.....kg
Total heat quantity for steam applications....MWh
Total heat quantity for all other applications.....kWh
Total standard volume.....(N)m3
Density.....kg/m3
Enthalpy.....kJ/kg

And with that making the unit correct since we are measuring heat and not steam and kWh applies to everything else unless it's steam which would then report MWh, sorry about that.

The documentation link: https://portal.endress.com/wa001/dla/5000429/0191/000/04/Ba00216ren_0211.pdf

Regards,
Ado