rscada/libmbus #100: Example to assing primary address

kamaxeon, 2016-04-14:
Hi !

This is not a bug, this is a question. I have not enough knowledge to do it. I have been reading the protocol, but I don't know how to assign primary address using  the secondary.

Somebody can help me ?

Thanks

Rico-van-Dongen, 2016-04-15:
Hi kamaxeon, 
You can not really assign addresses, however you can select the device. After selecting the device will be available at primary address 253. after you have finished reading the device you should deselect it with a SND_NKE message to make address 253 available for the next device. 
Selecting and deselecting is handled by the serial-request-data function so if you follow the steps in mbus-serial-request-data.c you should be able to find what you need. 
Rico

lategoodbye, 2016-04-15:
Hi kamaxeon,
i think you refers to MBDOC48.pdf chapter 6.4.2 Writing Data to a Slave. In theory you need to select a slave via secondary address and then send SND_UD message with primary address record ( data = 253 ). Unfortunately i've never give it a try neither i don't know which M-Bus slaves support that feature.

kamaxeon, 2016-04-15:
Hi again,

In this [pdf](http://www.janitza.com/download-manuals-current-devices.html?file=files/download/manuals/ECSEM-Series/Janitza-Technical-description-InBuilt-M-Bus-Protocol-en.pdf), point 3.7.2 is the instruction that I need, but I don't know how to do it :-(

The M-Bus slave support this feature, it has been tested with a [windows software](http://www.piigab.com/en/products/free-tools/m-bus-wizard/) and works like a charm.

mape1, 2016-05-06:
Hi,

I have a need for the same type of functionality that kamaxeon requests. I want to be able to assign primary addresses using libmbus.

mpseybold, 2018-03-02:
How about the example in pull request #118 ?
I haven't managed to compile this `patch', yet.

mpseybold, 2018-03-03:
I put the file mbus-serial-set-address in /libmbus/bin/ and added it to the Makefile.am of this folder.
With this ./build.sh compiles the programm.

I tested it with: Raspberry Pi B+ >> USB-RS232 >> Relay PW20 >> Modularis MBus Slave. https://www.wehrle.de/en/metering/remote-reading/modularis-modules/

Initially, the Module comes with primary address '0'. After './mbus-serial-set-address -d -b 2400 /dev/ttyUSB0 0 9' I receive 
>init_slaves: debug: sending init frame #1
[2018-03-03 21:07:23] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2018-03-03 21:07:23] SEND (005): 10 40 FD 3D 16
[2018-03-03 21:07:24] SEND (012): 68 06 06 68 53 00 51 01 7A 09 28 16

And './mbus-serial-request-data -b 2400 /dev/ttyUSB0 0' times out, as expected.
Whereas './mbus-serial-request-data -b 2400 /dev/ttyUSB0 9' returns the requested data.