rscada/libmbus #101: read more one meter ?

zhapwey, 2016-05-07:
hello
i am completely new to this.
I can read multiple counters in a facility or all at once?

lategoodbye, 2016-05-07:
I'm not sure that i understand your question. You need to select a meter via address (primary 0-255 or secondary XX:XX:XX:XX:XX:XX) and then request the data of this meter.

Please look at: http://www.m-bus.com/files/MBDOC48.PDF

zhapwey, 2016-05-07:
first thanks for your reply and sorry my english is very bad.

I read one by one all the devices without problem with mbus-serial-request-data.
I find them all without problems
my question is i can read multiples devices with a single command

lategoodbye, 2016-05-07:
There is a special primary address 0xFEh which makes all meter reply, but this would lead to bus collisions in case of multiple slaves. So according to your question the answer would be: no

closed 2016-05-16 by lategoodbye