rscada/libmbus #105: Pollustat E: Problems reading multi telegram

mape1, 2016-05-16:
I am trying to read out data from a Sensus Pollustat E. Using MiniCom I can see that the meter is reporting multiple telegrams but when I use libmbus to request data I only get a single telegram which seems to be the oldest one (using mbus-serial-request-data-multi-reply). If I reconfigure the meter using MiniCom to only report single telegrams I will get the correct data (the newest telegram). 

Am I misuing mbus-serial-request-data-multi-reply in some way or doesnt libmbus support multi telegram readouts for this meter?

lategoodbye, 2016-05-16:
Hi @mape1,
could you please write down your complete commandline of mbus-serial-request-data-multi-reply ?
Do you use the latest github version?
Could you please attach the output with parameter -d ?

mape1, 2016-05-19:
Hi @lategoodbye!

Sorry for the late response. I need some more time to be able to extract the information that you requested. I do not have direct access to the meter right now, but I will update the issue as soon as I have all relevant info for you.

delreich, 2016-05-20:
Workaround: use secondary addressing. Might need to use wildcards for the manufacturer/medium bit (i.e. the last 8 hex-digits).

The issue: the `mbus-serial-request-data*` tools currently doesn't handle `SND_NKE` and the FCB-bit correctly. `SND_NKE` is only ever sent to secondary address, and (initial) FCB-bit is fixed. Your Pollustat is stuck in a state where it previously sent the last frame in response to a request with a set FCB-bit, so when you next fire up `mbus-serial-request-data-multi-reply` and it sends a request with a set FCB-bit (without resetting first), it responds with the same frame.

Related: Issues #90, #97, and PR #95 (which I think might just make things worse...)

mape1, 2016-06-02:
Thanks for the advice, @delreich. I will try this out as soon as possible. 

Do you know if there are any plans currently to solve this issue?

lategoodbye, 2016-06-05:
Currently i don't have a testing environment. But fixes are always welcome.