rscada/libmbus #110: LandisGyr G350 48bit date/time field?

Rico-van-Dongen, 2016-07-20:
Hi, 
I'm working with a Landis+Gyr G350 gas meter and it seems to send a 48bit (6 byte) date-time structure. Libmbus only recognizes the 2 byte (F) and 4 byte (G) data types and I cannot find any documentation about this 6 byte structure. Does anyone know how to decode this field or what goes wrong in parsing of the data. 

This is (part of) the data messages that I get with the debug option of the mbus-serial-request-data function: 
68 40 40 68 08 01 72 58 20 08 12 E2 30 40 03 F1 00 00 00 // header 
 2F 2F 	// idle filler
 4C 13 	84 40 83 10 // gass value
 **46 6D** 	**00 00 09 14 27 00** // date-time
 0D 78 11 	34 31 38 35 30 32 38 30 32 31 39 35 37 31 30 30 47 // fabrication number (string)
 .......

Thanks.

lategoodbye, 2016-07-21:
@Rico-van-Dongen Could you please provide a complete and valid telegram incl. the date/time field?

Rico-van-Dongen, 2016-07-22:
@lategoodbye Below you will find the full frame. I marked the date time field as **bold**. My guess is that the date/time field is quite similar to the 4 byte structure with an extra byte for the seconds. This seems to result in plausible values but I don’t know for sure if this is correct and  how the time valid and summertime flags are coded. 
Thanks

mbus_frame_print: Dumping M-Bus frame [type 4, 70 bytes]: 68 40 40 68 08 01 72 58 20 08 12 E2 30 40 03 40 00 00 00 2F 2F 4C 13 92 40 83 10 **46 6D 00 00 08 16 27 00** 0D 78 11 34 31 38 35 30 32 38 30 32 31 39 35 37 31 30 30 47 89 40 FD 1A 01 01 FD 17 00 01 FD 67 0F 38 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>12082058</Id>
        <Manufacturer>LGB</Manufacturer>
        <Version>64</Version>
        <ProductName></ProductName>
        <Medium>Gas</Medium>
        <AccessNumber>64</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>10834092</Value>
        <Timestamp>2016-07-22T07:26:49</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>167873347584</Value>
        <Timestamp>2016-07-22T07:26:49</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Fabrication number</Unit>
        <Value>G0017591208205814</Value>
        <Timestamp>2016-07-22T07:26:49</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>1</Device>
        <Unit>Digital output (binary)</Unit>
        <Value>1</Value>
        <Timestamp>2016-07-22T07:26:49</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Error flags</Unit>
        <Value>0</Value>
        <Timestamp>2016-07-22T07:26:49</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Unrecognized VIF extension: 0x67</Unit>
        <Value>15</Value>
        <Timestamp>2016-07-22T07:26:49</Timestamp>
    </DataRecord>

</MBusData>

Rico-van-Dongen, 2016-07-22:
I found that the Itron G4 RF1 eV SW uses the same date – time structure. I have contacted Itron support and they responded very quickly. It turns out this is a Type I CP48 time struct. I will add the code to decode this today and share it here. 

[image]

Rico-van-Dongen, 2016-07-22:
Below you will find the changes I made. 
I’ve added the 6 byte structure to the mbus_data_tm_decode function (line 776) and the check for the vif 0x6D in the mbus_data_record_decode function (line 2527). I am not sure if other vifs like Start of tariff and date and time of battery change can also be send as Type I CP48 so I suggest to keep it restricted to 0x6D for now. 
Furthermore I changed the year coding to be from years since 1900. This allows the time library to convert the time struct to epoch times etc. 
@lategoodbye could you review and merge the code. 

[mbus-protocol.txt](https://github.com/rscada/libmbus/files/378135/mbus-protocol.txt)

lategoodbye, 2016-07-22:
@Rico-van-Dongen Thanks, but could you please create a unified patch against current master?

Rico-van-Dongen, 2016-07-22:
@lategoodbye Is this what you mean? I don't use git that much an I don’t think I have the permissions to commit here.  
I have created a .diff with SourceTree and renamed it to .txt to upload it here. 
[48bit datetime field_diff.txt](https://github.com/rscada/libmbus/files/378194/48bit.datetime.field_diff.txt)

lategoodbye, 2016-07-22:
@Rico-van-Dongen If you create a pull request then you don't need commit permission for this repo. If it's okay for you i'll create one based on your changes. In case you want to be mentioned as commit author i'll need your email address.

closed 2016-07-28 by lategoodbye