rscada/libmbus #112: Libtoolize and automake trouble

mitmacher, 2016-08-11:
Hi,
I'm having trouble installing the library on Raspberry Pi 3. 

I start with downloading the zip and unzip it, cd to the directory and then 

`autoheader && aclocal && libtoolize --ltdl --copy --force && automake --add-missing --copy && autoconf`

which should normally fix the bug, but for me it does not.

After this it also says:

Consider using 'AC_CONFIG_AUX_DIR([libtdl/config])' in configure.ac.

automake: warnings are treated as errors
/usr/share/automake-1.14/am/ltllibrary.am: warning: 'libmbus.la': linking libtool libraries using a non-POSIX
/usr/share/automake-1.14/am/ltllibrary.am: archiver requires 'AM_PROG_AR' in 'configure.ac'
mbus/Makefile.am16: while processing Libtool library 'libmbus.la'

Anyone can help me with these errors?

lategoodbye, 2016-08-11:
@mitmacher 

Does the patch from @jasp fix your issue?
https://github.com/keepfocus/libmbus/commit/111bbb71d4ef3eed8ecc36bf5353da16a401828c.patch

mitmacher, 2016-08-17:
Worked. Thank you very much.

closed 2016-08-17 by mitmacher