rscada/libmbus #113: Not able to find Meter - with mbus-serial-scan/mbus-serial-scan-secondary

derbodo, 2016-08-14:
Hello,

i'm having some trouble with getting libmbus to work.
Host for Libmbus : Ubuntu 16.04
Level Converter : HD67020-B2-10 from ADFWEB
Meter: GWF Water Meter

Issue looks like : https://github.com/rscada/libmbus/issues/24

The Solution marked there is changing the behavior due to the echo from the level converter:

I've tried all three options in the "mbus-protocol-auc.c" file at line 1523:

handle->purge_first_frame = MBUS_FRAME_PURGE_S2M;

The behavior doesn't change at all.

With the free software from Relay (MBSHEET) the meter can be found and read without any problems.

Maybe it's just a handling problem on my side.... any help is appreciated !

See also attached console output:

`mbus-serial-scan -d -b 2400 /dev/ttyUSB0
Scanning primary addresses:
0 [2016-08-14 15:42:57] SEND (005): 10 40 00 40 16
[2016-08-14 15:42:58] RECV (005): 10 40 00 40 16
1 [2016-08-14 15:42:58] SEND (005): 10 40 01 41 16
[2016-08-14 15:42:58] RECV (005): 10 40 01 41 16
2 [2016-08-14 15:42:58] SEND (005): 10 40 02 42 16
[2016-08-14 15:42:58] RECV (005): 10 40 02 42 16
3 [2016-08-14 15:42:58] SEND (005): 10 40 03 43 16
[2016-08-14 15:42:58] RECV (005): 10 40 03 43 16
4 [2016-08-14 15:42:58] SEND (005): 10 40 04 44 16
[2016-08-14 15:42:58] RECV (005): 10 40 04 44 16
5 [2016-08-14 15:42:58] SEND (005): 10 40 05 45 16
[2016-08-14 15:42:58] RECV (005): 10 40 05 45 16
`
Thank you for your help !

lategoodbye, 2016-08-14:
Does your Ubuntu run in a virtual machine or is it native one?

derbodo, 2016-08-14:
The Ubuntu is running inside a VM.
The MBSHEET on the Windows Machine is also running inside a VM on the same ESX Host.

With Option Search Address:

`Search baud rate 2400!
Slave primary search started!
TX-buffer:
10 40 FD 3D 16 
RX-buffer:

TX-buffer:
10 40 FF 3F 16 
RX-buffer:

TX-buffer:
10 40 01 41 16 
RX-buffer:
E5 
TX-buffer:
10 7B 01 7C 16 
RX-buffer:
68 1B 1B 68 08 01 72 58 63 77 15 E6 1E 3C 07 10 00 00 00 0C 78 58 63 77 15 0C 13 04 11 00 00 18 16 
*** Slave found at primary address 1`

With Option Search ID:

`Search baud rate 2400!
Slave ID search started!
TX-buffer:
10 40 FD 3D 16 
RX-buffer:

TX-buffer:
10 40 FF 3F 16 
RX-buffer:

0FFFFFFF
1FFFFFFF
*** Slave found at ID (secondary adress) 15776358
2FFFFFFF
Slave ID search finished!
`

derbodo, 2016-08-14:
Did i do the change to Frame purge at the correct location ?
Can i somehow verify that ?

Just want to make sure that i didn't make a fault.

lategoodbye, 2016-08-14:
@derbodo I don't think it's your fault. But there could be many reason for this behavior.

For instance VM plus USB to RS232 adapter changes runtime behavior. So it's possible that the responses are accidentially dropped.

Are you able to verify that the water meter response to libmbus over serial line?

derbodo, 2016-08-15:
Ok just checked it with an old Notebook that still has a serial port.
It works fine.
Switched from Serial Port to USB2Serial adapter.
Just works fine.
After that i just reinstalled the RPI2 an connected the USB2Serial adapter.
Works like a charm.

So the Problem is the ESXi host.
Not a problem related to libmbus - but has someone already successfully testet this within a linux VM ?

If possible i do not want to operate another PI beside the ESXi Server...
The RS485 Modbus Stick works just fine within the ESXi Ubuntu VM.....

derbodo, 2016-08-17:
Solution on my side is actually:

1. Use the USB Serial Adapter on a Raspberry
or
2. When using it within an ESX VM - do NOT use the Serial Adapter with a USB 3 Port - I'Ve now connected it to a USB 2 Hub and it works....

I'm closing this issue....

closed 2016-08-17 by derbodo