rscada/libmbus #116: AMT Aquametro CALEC MB

ado464, 2016-12-08:
Hello all, i am guessing my values are a bit off with the infinite decimals i got :P could someone please help me out

This is my output:
```
<MBusData>

    <SlaveInformation>
        <Id>4526402</Id>
        <Manufacturer>AMT</Manufacturer>
        <Version>128</Version>
        <ProductName>Aquametro CALEC MB</ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>156</AccessNumber>
        <Status>00</Status>
        <Signature>FFFF</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (100 Wh)</Unit>
        <Value>7620560</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (1e-1  m^3)</Unit>
        <Value>41011158</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (date)</Unit>
        <Value>2016-06-30</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Energy (100 Wh)</Unit>
        <Value>7329205</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>On time (hours)</Unit>
        <Value>95868</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (kW)</Unit>
        <Value>11174.887695</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow ( m^3/h)</Unit>
        <Value>305.355103</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (deg C)</Unit>
        <Value>115.947960</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (deg C)</Unit>
        <Value>83.715706</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Temperature Difference ( deg C)</Unit>
        <Value>32.232254</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2016-12-08T11:55:00</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (1e-1 Wh)</Unit>
        <Value>7620560125091</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="12">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (my m^3)</Unit>
        <Value>4101115816150</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="13">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>10.000000</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="14">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>1.000000</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="15">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Error flags</Unit>
        <Value>8192</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

    <DataRecord id="16">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (Wh)</Unit>
        <Value>1.135396</Value>
        <Timestamp>2016-12-08T09:59:36</Timestamp>
    </DataRecord>

</MBusData>
```

In hex:
```
init_slaves: debug: sending init frame #1
[2016-12-08 10:04:54] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2016-12-08 10:04:54] SEND (005): 10 40 FD 3D 16
[2016-12-08 10:04:54] SEND (005): 10 7B 01 7C 16
[2016-12-08 10:04:55] RECV (128): 68 7A 7A 68 08 01 72 02 64 52 04 B4 05 80 04 9D 00 FF FF 04 85 7D DA 47 74 00 04 15 E4 C8 71 02 42 6C 1E 26 44 85 7D B5 D5 6F 00 03 22 7C 76 01 05 2E 5B 9C 2F 46 05 3E 75 94 99 43 05 5B EB C9 E7 42 05 5F F8 55 A7 42 05 63 E6 E7 00 42 04 6D 00 0C 08 2C 06 02 C3 AF 60 4D EE 06 06 10 95 25 29 DF BA 03 05 93 28 00 00 20 41 05 93 29 00 00 80 3F 02 FD 17 00 20 05 83 33 3B 55 91 3F 93 16
mbus_frame_print: Dumping M-Bus frame [type 4, 128 bytes]: 68 7A 7A 68 08 01 72 02 64 52 04 B4 05 80 04 9D 00 FF FF 04 85 7D DA 47 74 00 04 15 E4 C8 71 02 42 6C 1E 26 44 85 7D B5 D5 6F 00 03 22 7C 76 01 05 2E 5B 9C 2F 46 05 3E 75 94 99 43 05 5B EB C9 E7 42 05 5F F8 55 A7 42 05 63 E6 E7 00 42 04 6D 00 0C 08 2C 06 02 C3 AF 60 4D EE 06 06 10 95 25 29 DF BA 03 05 93 28 00 00 20 41 05 93 29 00 00 80 3F 02 FD 17 00 20 05 83 33 3B 55 91 3F 93 16 
```

I can se that the temperature is way off with that many decimals so everything else must be off as well? Maybe even wrong values?

Thanks for all the help guys!

Best regards,
Ado

lategoodbye, 2016-12-08:
Hello @ado464,
could please name the affected datarecord ids which are wrong?
Are you able to cross check with a different M-Bus tool?
Did you use the latest version from github?

delreich, 2017-02-09:
As far as I can tell, there's no issue with the precision, at least not as far as the protocol goes. Possibly the meter is reporting more precise values than it actually has, but the values presented here match what it's sending. The non-integer values are all single-precision (32 bit) IEEE 754 floats.

There are some actual issues here though.
- The energy values in DR#0 and DR#3 are off by a factor of 1000 (VIFE 0x7D; "Multiplicative correction factor: 10³"). Compare DR#0 and DR#11.
- The energy value in DR#16 is actually "energy per (Kelvin * liter)" (VIFE 0x33).
- The volume values in DR#13 and DR#14 are actually "volume increment per input pulse on input channel 0/1" (VIFE 0x28 and 0x29).

[Pretty-printed](https://gist.github.com/delreich/18ff19130bd8a13fb60aa248dac9d7d7) frame dump:
```
Header:                 68 7A 7A 68 08 01 72
Slave Information:      02 64 52 04 B4 05 80 04 9D 00 FF FF
Data Records:
   DIB: 04                                 VIB: 85 7D
     Data: DA 47 74 00
   DIB: 04                                 VIB: 15
     Data: E4 C8 71 02
   DIB: 42                                 VIB: 6C
     Data: 1E 26
   DIB: 44                                 VIB: 85 7D
     Data: B5 D5 6F 00
   DIB: 03                                 VIB: 22
     Data: 7C 76 01
   DIB: 05                                 VIB: 2E
     Data: 5B 9C 2F 46
   DIB: 05                                 VIB: 3E
     Data: 75 94 99 43
   DIB: 05                                 VIB: 5B
     Data: EB C9 E7 42
   DIB: 05                                 VIB: 5F
     Data: F8 55 A7 42
   DIB: 05                                 VIB: 63
     Data: E6 E7 00 42
   DIB: 04                                 VIB: 6D
     Data: 00 0C 08 2C
   DIB: 06                                 VIB: 02
     Data: C3 AF 60 4D EE 06
   DIB: 06                                 VIB: 10
     Data: 95 25 29 DF BA 03
   DIB: 05                                 VIB: 93 28
     Data: 00 00 20 41
   DIB: 05                                 VIB: 93 29
     Data: 00 00 80 3F
   DIB: 02                                 VIB: FD 17
     Data: 00 20
   DIB: 05                                 VIB: 83 33
     Data: 3B 55 91 3F
Checksum and Stop:      93 16
```

lategoodbye, 2017-02-12:
I think it's time to use the new M-Bus decoding (mbus-protocol-aux.c) instead of the old implementation (mbus-protocol.c) for all the binaries.

    test/mbus_parse_hex -h <hex-file>

should decode it properly:

    <?xml version="1.0" encoding="ISO-8859-1"?>
    <MBusData>

    <SlaveInformation>
        <Id>4526402</Id>
        <Manufacturer>AMT</Manufacturer>
        <Version>128</Version>
        <ProductName>Aquametro CALEC MB</ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>157</AccessNumber>
        <Status>00</Status>
        <Signature>FFFF</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Wh</Unit>
        <Quantity>Energy</Quantity>
        <Value>762057000000.000000</Value>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3</Unit>
        <Quantity>Volume</Quantity>
        <Value>4101142.800000</Value>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>-</Unit>
        <Quantity>Time point (date)</Quantity>
        <Value>2016-06-30</Value>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Wh</Unit>
        <Quantity>Energy</Quantity>
        <Value>732920500000.000000</Value>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>s</Unit>
        <Quantity>On time</Quantity>
        <Value>345124800.000000</Value>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>W</Unit>
        <Quantity>Power</Quantity>
        <Value>11239088.867188</Value>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3/h</Unit>
        <Quantity>Volume flow</Quantity>
        <Value>307.159821</Value>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>°C</Unit>
        <Quantity>Flow temperature</Quantity>
        <Value>115.894371</Value>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>°C</Unit>
        <Quantity>Return temperature</Quantity>
        <Value>83.667908</Value>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>K</Unit>
        <Quantity>Temperature difference</Quantity>
        <Value>32.226463</Value>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>-</Unit>
        <Quantity>Time point (date &amp; time)</Quantity>
        <Value>2016-12-08T12:00:00</Value>
    </DataRecord>

    <DataRecord id="11">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Wh</Unit>
        <Quantity>Energy</Quantity>
        <Value>762057016518.700073</Value>
    </DataRecord>

    <DataRecord id="12">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3</Unit>
        <Quantity>Volume</Quantity>
        <Value>4101142.816149</Value>
    </DataRecord>

    <DataRecord id="13">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3</Unit>
        <Quantity>Volume</Quantity>
        <Value>0.010000</Value>
    </DataRecord>

    <DataRecord id="14">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3</Unit>
        <Quantity>Volume</Quantity>
        <Value>0.001000</Value>
    </DataRecord>

    <DataRecord id="15">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit></Unit>
        <Quantity>Error flags</Quantity>
        <Value>8192.000000</Value>
    </DataRecord>

    <DataRecord id="16">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Wh</Unit>
        <Quantity>Energy</Quantity>
        <Value>1.135414</Value>
    </DataRecord>

    </MBusData>

delreich, 2017-02-13:
I'm not sure I would call that "properly" decoded. It solves the issue with VIFE 0x7D, but still misrepresents data records 13, 14 and 16.