rscada/libmbus #119: Installing libmbus library on windows

siimonc, 2017-03-06:
Hi,

I tried to install and compile this library using MinGW and msys, only to realise that they do not support automake 1.12 and thus the libray cannot be compiled. I would like to remind other people like me intended to run the library on windows that it is not possible to use msys to compile the library.

I just want to know the error-free way to compile the library, **is ubuntu-version stable**? I notice that there is a difference in the source code for libmbus-0.8.0 and libmbus-master, I think libmbus-master is more up-to-date, thanks to all the active developers. I really appreciate the work you guys have put into this open-source library. Can you let me know **how to properly use autotools to build the library**? 

regards,
-Chen

lategoodbye, 2017-03-06:
Hi Chen,
without providing any useful information it's harder than necessary to help you.

Regarding to MinGW & msys:
What output did you get (libmbus should work with other versions of automake)?

Regarding to build on Ubuntu:
Please do not use libmbus-0.8.0 this release is nearly 5 years old.
Did you tried the build.sh?
What output did you get?

kamaxeon, 2017-03-06:
Hi siimonc,

I use libmbus with debian-jessie (raspberry pi), this is my setup:

apt-get -y install libtool automake 
cd /usr/src 
git clone https://github.com/rscada/libmbus.git 2>&1 
cd libmbus ./clean.sh 2>&1 
./build.sh 2>&1 make install  2>&1 
ln -s /usr/local/lib/libmbus.so.0 /usr/lib/libmbus.so.0 
chmod +s /usr/local/bin/mbus*

siimonc, 2017-03-07:
Hi Stefan,

I had installed MinGW and msys, added the directories "C:\..\MinGW\bin" and "C:\..\MinGW\msys\1.0\bin" to both local and system path in environment variables, as I was using a university computer which added some difficulties in having the admin rights. I followed the tutorial from mingw.org closely [link here: Getting_Started](http://mingw.org/wiki/Getting_Started), there was some confusion and that is to install msys separately in the wiki page which is outdated [Link here](http://mingw.org/wiki/MSYS). It took me some time to read through the comments to find out about this. 

I downloaded libmbus-master and extracted it manually and put them in the ..msys\1.0\home folder. I then ran the bat-file in msys, and navigated to the home folder where my extracted libmbus file is located. However, when I typed ./configure, it said something like the file could not be found. (Sorry I do not have the computer with msys with me now, so I could not rerun the code and provide the exact error lines). I realised there are difference in the source code in [GitHub master branch](https://github.com/rscada/libmbus) and in the [0.8.0 version](http://www.rscada.se/libmbus/index.php?lang=sv). As I investigate further, the master-branch is up-to-date, where changes were made from correcting the issues (I really appreaciate the work this community puts in), but how come I am not able to run ./configure in msys bat-file (like the problem I said before where the some files are missing)? 

I then downloaded the 0.8.0-version and extracted it in the msys home folder again, but this time using "tar -xf libmbus-0.8.0.tar.gz" and I was able to run ./configure after that. I could see that Makefile.in is produced, but there were errors after that when I ran _make_ and _make install_, the error was like some library is not defined so there are a whole bunch of variables which are not recognized. If I am not mistaken, it is the one in mbus/mbus-protocol.h. I thought the problem was that I was using automake 1.11 installed from the default MinGW graphical installer. I tried to install 1.12 but to no avail, as the whole msys is rooted to use automake 1.11. I tried to merge the latest version of configure.ac to the configure.ac in 0.8.0 version, which I was able to run ./configure, but it says _libltnl.config_ could not be found. I was really fed up of using msys, and that is when I saw #117 that it MIGHT be better to compile using cygwin. I am quite exhausted to try it in windows again and I thought it will be better to just use ubuntu directly with a VirtualBox. I have read your comments in the issues #42 and #117 and it makes me think that using msys is not the long-term solution. 

I am not from programming background but I have some understanding about programming. I just want to use this library to read and analyse the data from Hydrometer Sharky 773, which we bought in the university. 

I would really appreciate your help and sorry if it is too long to read. They are my personal encounters with msys and libmbus.

Cheers.
-Chen

siimonc, 2017-03-07:
Hi kamaxeon,

thank you, I will try it out if the ubuntu version does not work. 

Regards,
-siimonc

kamaxeon, 2017-03-07:
Hi simmonc,

What version of ubuntu are you testing?

siimonc, 2017-03-07:
it depends on the IT department of my univeristy, I cant tell you right now. Do you have some advice about Ubuntu? 

Regards.

kamaxeon, 2017-03-08:
Hi siimonc,

I've just tried on a ubuntu 16.04 (last LTS) with successfully result, very similar setup (only install make package):

apt-get -y install libtool automake make
cd /usr/src
git clone https://github.com/rscada/libmbus.git
cd libmbus
./clean.sh
./build.sh
make install
ln -s /usr/local/lib/libmbus.so.0 /usr/lib/libmbus.so.0
chmod +s /usr/local/bin/mbus*

Best regards,

siimonc, 2017-03-09:
hi kamaxeon,

thanks for your inputs! I will update the thread again once I have got back my working laptop.

Regards

kamaxeon, 2017-03-10:
Hi siimonc,

If you use ansible, here you have an ansible-role (https://galaxy.ansible.com/kamaxeon/libmbus/) for debian and ubuntu :-)

Best regards,

siimonc, 2017-03-12:
Hi kamaxeon,

I can now run Ubuntu 16.04.2 and followed your instructions above to compile the library. I could not execute the step ./build.sh and I got the following error message:

...:/usr/src/libmbus$ ./build.sh
Running autotools...
autom4te: cannot create autom4te.cache: No such file or directory
autoheader: '/usr/bin/autom4te' failed with exit status: 1
make: *** No targets specified and no makefile found.  Stop.

Would appreciate any inputs. Thanks!

lategoodbye, 2017-03-12:
@siimonc 

Could you please verify that you have also have autoconf and autotools-dev installed?

I've Ubuntu 16.04 and the build.sh script works.

siimonc, 2017-03-13:
@lategoodbye 

yes I have had them installed previously. Ran 

```
..:~$ sudo apt-get install autotools-dev
[sudo] password for siimonc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
autotools-dev is already the newest version (20150820.1).
autotools-dev set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```

```
..:~$ sudo apt-get install autoconf
Reading package lists... Done
Building dependency tree       
Reading state information... Done
autoconf is already the newest version (2.69-9).
autoconf set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```

the error message for autom4te.cache is still the same, 
```
..:/usr/src/libmbus$ ./build.sh
Running autotools...
**autom4te: cannot create autom4te.cache: No such file or directory**
autoheader: '/usr/bin/autom4te' failed with exit status: 1
make: *** No targets specified and no makefile found.  Stop.
```
any help would be great, thanks!
-siimonc

siimonc, 2017-03-13:
@lategoodbye 

I have solved the problem, I didnt have the permission to that directory and adding a sudo infront of ./build.sh solved the problem.

Thanks!
-siimonc

peacefulnuke, 2017-12-14:
Hi,

I was able to build libmbus on windows without problems. Cygwin is right tool for this task. Some extra packages needed to be installed. I'm sending what "cygcheck -c" returns. Not all of these packages are necessary to build library, but better to put everything here so you can see full working setup.
```
$ cygcheck -c
Cygwin Package Information
Package                    Version             Status
_autorebase                001006-1            OK
alternatives               1.3.30c-10          OK
autoconf                   13-1                OK
autoconf-archive           2017.09.28-1        OK
autoconf2.1                2.13-12             OK
autoconf2.5                2.69-3              OK
automake                   10-1                OK
automake1.10               1.10.3-3            OK
automake1.11               1.11.6-3            OK
automake1.12               1.12.6-3            OK
automake1.13               1.13.4-2            OK
automake1.14               1.14.1-3            OK
automake1.15               1.15.1-1            OK
automake1.4                1.4p6-11            OK
automake1.5                1.5-11              OK
automake1.6                1.6.3-12            OK
automake1.7                1.7.9-11            OK
automake1.8                1.8.5-11            OK
automake1.9                1.9.6-11            OK
base-cygwin                3.8-1               OK
base-files                 4.2-4               OK
bash                       4.4.12-3            OK
binutils                   2.29-1              OK
bzip2                      1.0.6-3             OK
ca-certificates            2.16-1              OK
coreutils                  8.26-2              OK
csih                       0.9.9-1             OK
cygrunsrv                  1.62-1              OK
cygutils                   1.4.16-2            OK
cygwin                     2.9.0-3             OK
cygwin-debuginfo           2.9.0-3             OK
cygwin-devel               2.9.0-3             OK
cygwin32-libtool           2.4.6-3             OK
dash                       0.5.9.1-1           OK
diffutils                  3.5-2               OK
editrights                 1.03-1              OK
file                       5.30-1              OK
findutils                  4.6.0-1             OK
gawk                       4.2.0-1             OK
gcc-core                   6.4.0-4             OK
gcc-g++                    6.4.0-4             OK
gcc-tools-epoch1-autoconf  2.59-2              OK
gcc-tools-epoch1-automake  1.9.6-2             OK
gcc-tools-epoch2-autoconf  2.64-2              OK
gcc-tools-epoch2-automake  1.11.6-1            OK
getent                     2.18.90-4           OK
git                        2.15.1-1            OK
grep                       3.0-2               OK
groff                      1.22.3-1            OK
gzip                       1.8-1               OK
hostname                   3.13-1              OK
info                       6.5-2               OK
ipc-utils                  1.0-2               OK
less                       487-1               OK
libargp                    20110921-3          OK
libatomic1                 6.4.0-4             OK
libattr1                   2.4.46-1            OK
libblkid1                  2.25.2-2            OK
libbz2_1                   1.0.6-3             OK
libcom_err2                1.42.12-2           OK
libcrypt0                  2.1-1               OK
libcurl4                   7.56.1-1            OK
libdb5.3                   5.3.28-2            OK
[... 85 lines cut ...]
publicsuffix-list-dafsa    20171028-1          OK
rebase                     4.4.2-1             OK
rsync                      3.1.2-1             OK
run                        1.3.4-2             OK
sed                        4.4-1               OK
tar                        1.29-1              OK
terminfo                   6.0-12.20171125     OK
texinfo                    6.5-2               OK
tzcode                     2017c-1             OK
tzdata                     2017c-1             OK
util-linux                 2.25.2-2            OK
vim-minimal                8.0.1376-1          OK
w32api-headers             5.0.3-1             OK
w32api-runtime             5.0.3-1             OK
which                      2.20-2              OK
windows-default-manifest   6.4-1               OK
xz                         5.2.3-1             OK
zlib0                      1.2.11-1            OK
```
To build library just type
```
./clean.sh
./build.sh
make install
```

Apollon77, 2018-02-27:
I currently try to build a nodejs native module with libmbus and therefor also I also try to get it compile and working on Windows ... TCP is already working ... Serial is next.
And I focussed on the lib itself for now ... so binaries are still "untouched".

But I could need support to create the needed msvc makefiles and such ... So if someone wants to join with msvc or at all "compiler know how" (because I'm not really the C/C++ guy) ... contact me (ingo@fischer-ka.de).

Current (more experimental) WIP status: https://github.com/Apollon77/libmbus/tree/build-windows

Apollon77, 2018-03-05:
Got serial and tcp working on windows and also release a first version of the nodejs native library to npm based on 
https://github.com/Apollon77/libmbus/tree/build-windows

Still a lot of work to do in order to create a PR :-) na yhelp would be great!

Aali1364, 2021-02-16:
> سلام siimonc ،
> 
> من از libmbus با debian-jessie (تمشک پی) استفاده می کنم ، این تنظیمات من است:
> 
> apt-get -y نصب libtool automake
> cd / usr / src
> git clone https://github.com/rscada/libmbus.git
> cd libmbus
> ./clean.sh
> ./build.sh
> نصب
> ln -s / usr / local / lib / libmbus.so.0 /usr/lib/libmbus.so.0
> chmod + s / usr / local / bin / mbus *

> lategoodbye
> 
> من این مشکل را برطرف کردم ، من مجوز آن فهرست را نداشتم و با اضافه کردن یک sudo در مقابل ./build.sh مشکل را حل کردم.
> 
> با تشکر!
> -siimonc

hi

Aali1364, 2021-02-16:
> Cygwin Package Information

Hello, can you simply explain what I should do in Windows or Ubuntu?

Aali1364, 2021-02-16:
> سریال و tcp روی ویندوز کار می کند و همچنین اولین نسخه از کتابخانه بومی nodejs را بر اساس https://github.com/Apollon77/libmbus/tree/build-windows به npm منتشر می
> [کنید](https://github.com/Apollon77/libmbus/tree/build-windows)
> 
> هنوز کارهای زیادی برای ایجاد یک روابط عمومی انجام می شود :-) na yhelp بسیار عالی خواهد بود!

Hello, should this be done inside Visual Studio?