rscada/libmbus #121: Hardcoded timeout not sufficiently long

ree2k, 2017-03-19:
libmbus-0.8.0:
the timeout should be a CLI parameter.

the hardcoded, default timeout of mbus-serial-....  is not sufficiently long for some meters (scanning and receiving) 
Tested with the PW20 M-Bus Master from Relay and the water meter module 
https://messhelden.com/shop/m-bus-modul-fuer-wasserzaehler/

The same hardware constellation works using the bulky java mbus library:
https://www.openmuc.org/m-bus/

THX for your efforts!

lategoodbye, 2017-03-19:
Those timeouts are calculated based on the M-Bus Spec, so force the user to specify isn't a nice solution. 

Beside the fact the 0.8.0 release is pretty, what makes you sure that the issue is the wrong timeout?
Which commands of libmbus did you use?
What was the output?
Did you use a USB to serial adapter?

ree2k, 2017-03-19:
Thx for the quick reply. The Setting:
Raspberry Pi B+ <=> FTDI USB to RS232 <=> Relay PW20 <=> Modularis M-Bus Module
https://www.wehrle.de/en/metering/remote-reading/modularis-modules/
The meter is directly connected to the mbus master - wire length is not the issue.

I used the binary mbus-serial-scan and did not find a slave on any primary address.
I used the script read-meter.sh from jMBus and did not find a slave on any primary address, too.

However, jMBus allows to overwrite the default scan timeout of 1000ms using the -t switch.
-t 2000 did successfully discover the meter.
The same for read-meter.sh , which only succeeds after overwriting the default 500ms timeout with -t 2000.

I didn't manage to try the current master of libmbus tough. The configure.ac file was not compatible with my version of autotools :(

lategoodbye, 2017-03-19:
Okay, at which baudrate is the Modularis M-Bus Module?
Do you have the issues with autotools from Raspbian?

ree2k, 2017-03-19:
The meter runs on 2400 bauds.
Yes, it's Raspbian Jessie.

$ cat /etc/issue
Raspbian GNU/Linux 8 \n \l

$ autoconf -V
autoconf (GNU Autoconf) 2.69

$ dpkg -l | grep auto
ii  autoconf                         2.69-8                     all          automatic configure script builder
ii  automake                         1:1.14.1-4+deb8u1          all          Tool for generating GNU Standards-compliant Makefiles
ii  autopoint                        0.19.3-2                   all          The autopoint program from GNU gettext
ii  autotools-dev                    20140911.1                 all          Update infrastructure for config.{guess,sub} files
ii  dh-autoreconf                    12~bpo8+1                  all          debhelper add-on to call autoreconf and clean up after the build

The autoconf "issue" looks like :
$ git clone https://github.com/rscada/libmbus.git
~/libmbus $ autoconf
configure.ac:15: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:17: error: possibly undefined macro: AM_PROG_LIBTOOL

lategoodbye, 2017-04-07:
I will try to find timeout values which should fit also in USB to serial adapter cases (common case nowadays).

lategoodbye, 2017-04-07:
@ree2k I adjusted the timeouts in this branch, please try:

https://github.com/rscada/libmbus/tree/serial-timeout

ree2k, 2017-04-08:
sry for the late reply. 
i) I did not use CLI interface properly. Both, 0.8.0 and the master, succeed in scanning via the USB-RS232 Adapter, if set up with the correct baud rate.

ii) However, requesting data from the Modularis module still times out ( using the correct baud rate).
Issuing the same CLI command right after the first attemt timed out does in fact retrieve the m-bus telegram.

iii) I will try your branch later today. Thx already.

lategoodbye, 2018-03-20:
I assume this isn't an issue anymore.

closed 2018-03-20 by lategoodbye

mpseybold, 2018-03-21:
Yes, it's not an issue anymore.
The current master works quite nice with a recent debian stretch and kernel.

My weird observation was somehow a mix of not properly using the library, old kernel  usb tty drivers and the automatic baudrate negotiation behaviour of the modularis module.
Sry for the inconvenience and thx again for the branch.