rscada/libmbus #123: DataRecod time issue

mmuruev, 2017-03-29:
`    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (date)</Unit>
        <Value>2127-15-31</Value>
        <Timestamp>2017-03-29T12:18:07</Timestamp>
    </DataRecord>
`
Value far in the future. Seems like somethere offset is wrong. It happens on in this place.

lategoodbye, 2017-03-29:
@mmuruev This bug report isn't helpful.
Please write down at least the following information:
Did you use the latest libmbus version from Github?
Which platform do you use for libmbus?
What was your command to get this output?
Do you know the identification / name of the meter?
Do you know the expected value than 2127-15-31?
Please try to enable the debug option of the used command (usually -d) and attach the output here.

mmuruev, 2017-03-29:
Yes lattest master.
Binary exchange.
[2017-03-29 12:18:06] SEND (005): 10 40 FD 3D 16
[2017-03-29 12:18:06] SEND (017): 68 0B 0B 68 73 FD 52 79 68 99 09 65 32 D8 07 BB 16
[2017-03-29 12:18:06] RECV (001): E5
[2017-03-29 12:18:07] SEND (005): 10 5B FD 58 16
[2017-03-29 12:18:07] RECV (066): 68 3C 3C 68 08 FD 72 79 68 99 09 65 32 D8 07 12 88 00 00 04 6D 0D 06 0C 2A 0C 13 83 01 00 00 4C 13 00 00 00 00 42 6C FF FF 32 6C 0D 2A 02 BB 56 00 00 06 FD 0C D8 00 07 00 5B 02 02 FA 3D 00 01 DF 16

Manufacturer 12901 water meter.

I belive all this <DataRecord> mostly nonsense..It looks like 'Statistics' fields. But in the code.. I cant see group by it.. I belive all part after 'header' in the code looks different. 'Albatros ID'  "Network index' just not present.  all values after splits only on dib and vib. Don't get idea on this. Why so complicated we have offsets for fields for decision which is which, but this one hide in list of records.

lategoodbye, 2017-03-30:
Okay, this is a step forward. But there are some misunderstandings.
Please provide me the name of the meter, for example Sontex Supercal 531  (no M-Bus data which aren't helpful) or even better a link to the meter manual as an PDF file.
Also i need to know your application platform, for example Ubuntu 16.04 Intel x86.

Only for comparison the output of mbus_parse_hex:

```
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>9996879</Id>
        <Manufacturer>LSE</Manufacturer>
        <Version>216</Version>
        <ProductName></ProductName>
        <Medium>Water</Medium>
        <AccessNumber>18</AccessNumber>
        <Status>88</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>-</Unit>
        <Quantity>Time point (date &amp; time)</Quantity>
        <Value>2016-10-12T06:13:00</Value>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3</Unit>
        <Quantity>Volume</Quantity>
        <Value>0.183000</Value>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>m^3</Unit>
        <Quantity>Volume</Quantity>
        <Value>0.000000</Value>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>-</Unit>
        <Quantity>Time point (date)</Quantity>
        <Value>2127-15-31</Value>
    </DataRecord>

    <DataRecord id="4">
        <Function>Value during error state</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>-</Unit>
        <Quantity>Time point (date)</Quantity>
        <Value>2016-10-13</Value>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>m^3/h</Unit>
        <Quantity>Volume flow</Quantity>
        <Value>0.000000</Value>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit></Unit>
        <Quantity>Device type</Quantity>
        <Value>2589865738456.000000</Value>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit></Unit>
        <Quantity>Bus Address</Quantity>
        <Value>256.000000</Value>
    </DataRecord>

</MBusData>
```

mmuruev, 2017-03-30:
Platform ubuntu 16.10 x64,
But I cant give you meter model I don't have it, as manual as well I have only 'official' utility generated correct values for compare and this dump. If it's not enough.. not too much here I can do. May be I can reverse this, but driver internal 'record' structure seems weird for me.

lategoodbye, 2017-03-30:
Those outputs would be also very helpful. Please attach them.

mmuruev, 2017-04-02:
Investigated. Really whey don't. The data presented here contain something different from program. So first for access I need put password in the mbus device! without it access denided, but it shows in the lib as Unknown Error. Not sure that such functionality supported. Pass dended by LONG FRAME.
` /// 						   P  P	  P	 P	       Ac P  P  P  P
 68 16 16 68 73 00 51 04 FD 11 A2 75 1F 00 04 FD   12 A2 75 1F 00 0F 5F 02 01 09 CF 16   `
Which looks like this. Any suggestions for create the access and pass field?

lategoodbye, 2017-04-07:
Sorry, i don't have access to the official spec. Or even worse this is vendor specific.

mmuruev, 2017-04-11:
Investigated Date issue. Date and Time field seems fine, but just Date. Produces maximum possible year. Because is has 1 in all bites. I'm suspecting that data range is incorrect.

fosterbuster, 2018-04-23:
According to the standard:

"For compatibility with old meters with a circular two digit date it is recommended to consider in any
master software the years “00” to “80” as the years 2000 to 2080. A value of FFh in both bytes (that
means FFh FFh) shall be interpreted as invalid."

If we follow the standard a year-value of FF,FF will ultimately yield 127.
You could then argue, that the correct parsing should be 2027. But none-the-less the issue here is most likely yhat the function should indicate error state.

closed 2026-10-17 by lategoodbye