rscada/libmbus #124: Timeout problem for 2400 baud

btrhad, 2017-04-07:
I am using libmbus on a Beaglebone black and had no problems in retrieving data from various meter types. Until I encountered a GWF UniCocoder water meter. This meter is fixed at 2400 baud. I was able to do an address scan, but an `serial-request-data` at 2400 baud always failed. Checking with a scope I found the device was sending a correct response, but the Beaglebone as master was not able to parse the response.
I found out the Timeout for 2400 baud in `mbus_serial_set_baudrate()` was just a little bit to short. 
Changing `handle-t.c.c_cc[VTIME]` from 2 to 3 solved the problem.

lategoodbye, 2017-04-07:
@btrhad I want to know more about your setup (starting from Beaglebone until M-Bus wire).
What kind of M-Bus master do you use?
Do you use Linux on the Beaglebone?

btrhad, 2017-04-07:
Hi Stefan,

We use beaglebone black (rev C), with this cape:  http://elinux.org/Apli_Hvac_cape (mbus/modbus) wich contains this mbus master interface:  http://www.solvimus.de/smart-metering/m-bus-oem-master/
We use linux 4.4.30-ti-r64 kernel with debian 8 (Jessie) this is the beagleboard.org shipped default.
And we add rscada/libmbus  software.

Please let me know if I need to provide more info

Met vriendelijke groet, With kind regards,
Betronic BV

Henk Dijkstra
Manager Connectivity Solutions
ü Please<http://www.betronic.nl/jahia/Jahia/lang/en/pid/153> consider the environment before printing this e-mail

lategoodbye, 2017-04-07:
@btrhad Could you please try this branch (which contains your suggestion):

https://github.com/rscada/libmbus/tree/serial-timeout

btrhad, 2017-04-07:
I can do that next week, don't have The module now at hand. I Will inform you of The result. Ps i also noted that The First req-ud2 starts with The fcb bit cleared, while all other mbus master sw starts with this bit set. Does this sound familiar?

Verstuurd vanaf mijn iPhone

https://github.com/rscada/libmbus/tree/serial-timeout

lategoodbye, 2017-04-07:
I created a branch to fix FCB handling:

https://github.com/rscada/libmbus/tree/fix-fcb-handling

but didn't get positive feedback. In case there is a real issue please feel free to open a new issue or at least try this branch, too.

btrhad, 2017-04-10:
Hi Stefan,

This branch works fine with all the mbus meters I have in my possession (at 9600 baud)  including the 2400 baud GWF meter.
Secondary address scan now also works reliable (which was not the case with libmbus release 0.8.0)

I will monitor for the rest of the week, to check reliability.

Met vriendelijke groet, With kind regards,
Betronic BV

Henk Dijkstra
Manager Connectivity Solutions
ü Please<http://www.betronic.nl/jahia/Jahia/lang/en/pid/153> consider the environment before printing this e-mail

https://github.com/rscada/libmbus/tree/serial-timeout

lategoodbye, 2017-04-16:
I merged the serial-timeout branch into the master.

closed 2017-04-27 by lategoodbye