rscada/libmbus #125: Schneider iEM3135 More than 25 values

ado464, 2017-06-09:
Hello i have a mbus meter iEM3135, i am trying to read value 32 but the output only goes to 25.

Hopefully this is a quick fix :) Thanks for all the help guys! 

Here is the output:

<pre>
init_slaves: debug: sending init frame #1
[2017-06-09 07:44:28] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2017-06-09 07:44:28] SEND (005): 10 40 FD 3D 16
[2017-06-09 07:44:28] SEND (005): 10 5B 02 5D 16
[2017-06-09 07:44:29] RECV (250): 68 F4 F4 68 08 02 72 62 30 31 03 A3 4C 15 02 16 00 00 00 0D FD 0A 12 63 69 72 74 63 65 6C 45 20 72 65 64 69 65 6E 68 63 53 0D FD 0C 08 20 35 33 31 33 4D 45 69 0D FD 0E 07 37 30 30 2E 33 2E 31 03 FD 17 00 00 00 05 FD DC FF 01 BE C6 08 42 05 FD DC FF 02 51 F3 0B 42 05 FD DC FF 03 11 3D 09 42 05 FD DC FF 00 0B FD 09 42 05 FD C9 FF 05 AA 6A C7 43 05 FD C9 FF 06 AC FA C7 43 05 FD C9 FF 07 8E 3C C7 43 05 FD C9 FF 08 C0 B2 C7 43 05 FD C9 FF 01 C7 DD 67 43 05 FD C9 FF 02 3E 0E 65 43 05 FD C9 FF 03 C9 E1 66 43 05 FD C9 FF 04 EF 99 66 43 05 AE FF 01 A0 ED C1 40 05 AE FF 02 2E 8D C3 40 05 AE FF 03 03 BB C0 40 05 2E 74 8D 91 41 85 40 2E CB 36 76 41 85 80 40 2E 02 A2 BE 41 05 FF 0A 3E 76 43 3F 05 FF 0B 64 F8 47 42 07 03 32 30 FD 01 00 00 00 00 1F C1 16
mbus_frame_print: Dumping M-Bus frame [type 4, 250 bytes]: 68 F4 F4 68 08 02 72 62 30 31 03 A3 4C 15 02 16 00 00 00 0D FD 0A 12 63 69 72 74 63 65 6C 45 20 72 65 64 69 65 6E 68 63 53 0D FD 0C 08 20 35 33 31 33 4D 45 69 0D FD 0E 07 37 30 30 2E 33 2E 31 03 FD 17 00 00 00 05 FD DC FF 01 BE C6 08 42 05 FD DC FF 02 51 F3 0B 42 05 FD DC FF 03 11 3D 09 42 05 FD DC FF 00 0B FD 09 42 05 FD C9 FF 05 AA 6A C7 43 05 FD C9 FF 06 AC FA C7 43 05 FD C9 FF 07 8E 3C C7 43 05 FD C9 FF 08 C0 B2 C7 43 05 FD C9 FF 01 C7 DD 67 43 05 FD C9 FF 02 3E 0E 65 43 05 FD C9 FF 03 C9 E1 66 43 05 FD C9 FF 04 EF 99 66 43 05 AE FF 01 A0 ED C1 40 05 AE FF 02 2E 8D C3 40 05 AE FF 03 03 BB C0 40 05 2E 74 8D 91 41 85 40 2E CB 36 76 41 85 80 40 2E 02 A2 BE 41 05 FF 0A 3E 76 43 3F 05 FF 0B 64 F8 47 42 07 03 32 30 FD 01 00 00 00 00 1F C1 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>3313062</Id>
        <Manufacturer>SEC</Manufacturer>
        <Version>21</Version>
        <ProductName></ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>22</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer (as in fixed header)</Unit>
        <Value>Schneider Electric</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Model / Version</Unit>
        <Value>iEM3135 </Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Firmware version</Unit>
        <Value>1.3.007</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Error flags</Unit>
        <Value>0</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> A</Unit>
        <Value>34.194084</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> A</Unit>
        <Value>34.987614</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> A</Unit>
        <Value>34.309635</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> A</Unit>
        <Value>34.497112</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>398.833313</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>399.958374</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>398.473083</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>399.396484</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="12">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>231.866318</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="13">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>229.055634</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="14">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>230.881973</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="15">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit> V</Unit>
        <Value>230.601303</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="16">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (kW)</Unit>
        <Value>6.060257</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="17">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (kW)</Unit>
        <Value>6.110984</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="18">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (kW)</Unit>
        <Value>6.022829</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="19">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (kW)</Unit>
        <Value>18.194069</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="20">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>1</Device>
        <Unit>Power (kW)</Unit>
        <Value>15.388377</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="21">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>2</Device>
        <Unit>Power (kW)</Unit>
        <Value>23.829105</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="22">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0.763523</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="23">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>49.992569</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="24">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (Wh)</Unit>
        <Value>33370162</Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

    <DataRecord id="25">
        <Function>More records follow</Function>
        <Value></Value>
        <Timestamp>2017-06-09T07:44:29</Timestamp>
    </DataRecord>

</MBusData>
</pre>

best regards,
Ado

lategoodbye, 2017-06-09:
Which application did you use?

ado464, 2017-06-09:
The latest libmbus version, i just did a git pull and recompiled and installed it, i get the same output :/ any way i could help out with more info? :)

best regards,
Ado

lategoodbye, 2017-06-09:
Let me rephrase my question. Did you tried mbus-serial-request-data-multi-reply ?

ado464, 2017-06-09:
Wow i must be an idiot, i got up to 78 values now! Now to check if it's correct haha. Amazing man, amazing

```
mbus-serial-request-data-multi-reply -d -b 2400 /dev/ttyUSB0 2

init_slaves: debug: sending init frame #1
[2017-06-09 08:35:10] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2017-06-09 08:35:11] SEND (005): 10 40 FD 3D 16
[2017-06-09 08:35:11] SEND (005): 10 7B 02 7D 16
[2017-06-09 08:35:12] RECV (250): 68 F4 F4 68 08 02 72 62 30 31 03 A3 4C 15 02 50 00 00 00 0D FD 0A 12 63 69 72 74 63 65 6C 45 20 72 65 64 69 65 6E 68 63 53 0D FD 0C 08 20 35 33 31 33 4D 45 69 0D FD 0E 07 37 30 30 2E 33 2E 31 03 FD 17 00 00 00 05 FD DC FF 01 CD FC 07 42 05 FD DC FF 02 5D 67 0B 42 05 FD DC FF 03 9E 9D 08 42 05 FD DC FF 00 ED 55 09 42 05 FD C9 FF 05 03 29 C8 43 05 FD C9 FF 06 64 CE C8 43 05 FD C9 FF 07 AA DF C7 43 05 FD C9 FF 08 B0 47 C8 43 05 FD C9 FF 01 AF 64 68 43 05 FD C9 FF 02 60 E9 65 43 05 FD C9 FF 03 B3 83 67 43 05 FD C9 FF 04 EC 45 67 43 05 AE FF 01 AC B6 C0 40 05 AE FF 02 E3 B4 C2 40 05 AE FF 03 BE A7 BF 40 05 2E D4 C4 90 41 85 40 2E CF 24 78 41 85 80 40 2E 82 FB BE 41 05 FF 0A A1 9E 42 3F 05 FF 0B 3F CA 47 42 07 03 0A 6C FD 01 00 00 00 00 1F 77 16
[2017-06-09 08:35:12] SEND (005): 10 5B 02 5D 16
[2017-06-09 08:35:14] RECV (252): 68 F6 F6 68 08 02 72 62 30 31 03 A3 4C 15 02 51 00 00 00 07 83 FF 09 11 00 00 00 00 00 00 00 87 40 03 98 7D 02 02 00 00 00 00 87 40 83 FF 09 00 00 00 00 00 00 00 00 04 ED FF 0C 00 00 01 01 07 83 FF 0D 0A 6C FD 01 00 00 00 00 87 40 83 FF 0D 98 7D 02 02 00 00 00 00 07 83 FF 01 62 13 A9 00 00 00 00 00 07 83 FF 02 80 08 AD 00 00 00 00 00 07 83 FF 03 38 50 A7 00 00 00 00 00 04 ED FF 0E 00 00 01 01 07 FD 61 00 00 00 00 00 00 00 00 03 FF 10 00 00 00 87 10 03 00 00 00 00 00 00 00 00 87 20 03 00 00 00 00 00 00 00 00 87 30 03 00 00 00 00 00 00 00 00 87 80 10 03 00 00 00 00 00 00 00 00 04 6D 21 09 29 26 03 FF 2C 64 00 00 03 FF 2D 00 00 00 05 FF 2E 00 00 C8 42 05 FF 2F 00 00 FA 43 03 FF 30 00 00 00 03 FD 1B 00 00 00 02 FF 32 00 00 03 FD 1A FF FF 00 1F A8 16
[2017-06-09 08:35:14] SEND (005): 10 7B 02 7D 16
[2017-06-09 08:35:15] RECV (204): 68 C6 C6 68 08 02 72 62 30 31 03 A3 4C 15 02 52 00 00 00 02 FF 34 00 00 05 FF 35 00 00 80 3F 02 FF 36 00 00 02 FF 37 00 00 02 FF 38 00 00 04 ED FF 39 00 00 01 01 05 FF 3A 00 00 C0 FF 06 FF 20 01 67 A7 00 00 00 03 FF 21 03 00 00 03 FF 22 04 00 00 03 FF 23 0B 00 00 03 FF 24 32 00 00 05 03 7F 69 02 47 05 83 FF 09 96 43 8B 3C 85 40 03 AB B5 03 47 85 40 83 FF 09 00 00 00 00 05 83 FF 0D 7F 69 02 47 85 40 83 FF 0D AB B5 03 47 05 83 FF 01 34 22 2D 46 05 83 FF 02 A4 2F 31 46 05 83 FF 03 36 54 2B 46 05 FD 61 00 00 00 00 85 10 03 00 00 00 00 85 20 03 00 00 00 00 85 30 03 00 00 00 00 85 80 10 03 00 00 00 00 0F BA 16
mbus_frame_print: Dumping M-Bus frame [type 4, 250 bytes]: 68 F4 F4 68 08 02 72 62 30 31 03 A3 4C 15 02 50 00 00 00 0D FD 0A 12 63 69 72 74 63 65 6C 45 20 72 65 64 69 65 6E 68 63 53 0D FD 0C 08 20 35 33 31 33 4D 45 69 0D FD 0E 07 37 30 30 2E 33 2E 31 03 FD 17 00 00 00 05 FD DC FF 01 CD FC 07 42 05 FD DC FF 02 5D 67 0B 42 05 FD DC FF 03 9E 9D 08 42 05 FD DC FF 00 ED 55 09 42 05 FD C9 FF 05 03 29 C8 43 05 FD C9 FF 06 64 CE C8 43 05 FD C9 FF 07 AA DF C7 43 05 FD C9 FF 08 B0 47 C8 43 05 FD C9 FF 01 AF 64 68 43 05 FD C9 FF 02 60 E9 65 43 05 FD C9 FF 03 B3 83 67 43 05 FD C9 FF 04 EC 45 67 43 05 AE FF 01 AC B6 C0 40 05 AE FF 02 E3 B4 C2 40 05 AE FF 03 BE A7 BF 40 05 2E D4 C4 90 41 85 40 2E CF 24 78 41 85 80 40 2E 82 FB BE 41 05 FF 0A A1 9E 42 3F 05 FF 0B 3F CA 47 42 07 03 0A 6C FD 01 00 00 00 00 1F 77 16 
mbus_frame_print: Dumping M-Bus frame [type 4, 252 bytes]: 68 F6 F6 68 08 02 72 62 30 31 03 A3 4C 15 02 51 00 00 00 07 83 FF 09 11 00 00 00 00 00 00 00 87 40 03 98 7D 02 02 00 00 00 00 87 40 83 FF 09 00 00 00 00 00 00 00 00 04 ED FF 0C 00 00 01 01 07 83 FF 0D 0A 6C FD 01 00 00 00 00 87 40 83 FF 0D 98 7D 02 02 00 00 00 00 07 83 FF 01 62 13 A9 00 00 00 00 00 07 83 FF 02 80 08 AD 00 00 00 00 00 07 83 FF 03 38 50 A7 00 00 00 00 00 04 ED FF 0E 00 00 01 01 07 FD 61 00 00 00 00 00 00 00 00 03 FF 10 00 00 00 87 10 03 00 00 00 00 00 00 00 00 87 20 03 00 00 00 00 00 00 00 00 87 30 03 00 00 00 00 00 00 00 00 87 80 10 03 00 00 00 00 00 00 00 00 04 6D 21 09 29 26 03 FF 2C 64 00 00 03 FF 2D 00 00 00 05 FF 2E 00 00 C8 42 05 FF 2F 00 00 FA 43 03 FF 30 00 00 00 03 FD 1B 00 00 00 02 FF 32 00 00 03 FD 1A FF FF 00 1F A8 16 
mbus_frame_print: Dumping M-Bus frame [type 4, 204 bytes]: 68 C6 C6 68 08 02 72 62 30 31 03 A3 4C 15 02 52 00 00 00 02 FF 34 00 00 05 FF 35 00 00 80 3F 02 FF 36 00 00 02 FF 37 00 00 02 FF 38 00 00 04 ED FF 39 00 00 01 01 05 FF 3A 00 00 C0 FF 06 FF 20 01 67 A7 00 00 00 03 FF 21 03 00 00 03 FF 22 04 00 00 03 FF 23 0B 00 00 03 FF 24 32 00 00 05 03 7F 69 02 47 05 83 FF 09 96 43 8B 3C 85 40 03 AB B5 03 47 85 40 83 FF 09 00 00 00 00 05 83 FF 0D 7F 69 02 47 85 40 83 FF 0D AB B5 03 47 05 83 FF 01 34 22 2D 46 05 83 FF 02 A4 2F 31 46 05 83 FF 03 36 54 2B 46 05 FD 61 00 00 00 00 85 10 03 00 00 00 00 85 20 03 00 00 00 00 85 30 03 00 00 00 00 85 80 10 03 00 00 00 00 0F BA 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>3313062</Id>
        <Manufacturer>SEC</Manufacturer>
        <Version>21</Version>
        <ProductName></ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>80</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0" frame="0">
        <Function>Instantaneous value</Function>
[... cut ...]
    <DataRecord id="77" frame="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>4</Tariff>
        <Device>0</Device>
        <Unit>Energy (Wh)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2017-06-09T08:35:15</Timestamp>
    </DataRecord>

    <DataRecord id="78" frame="2">
        <Function>Manufacturer specific</Function>
        <Value></Value>
        <Timestamp>2017-06-09T08:35:15</Timestamp>
    </DataRecord>

</MBusData>
```

best regards,
Ado

ado464, 2017-06-09:
Hmmm sometimes it returns 26 values, somtimes only 52 and sometimes 78. I have 2 more mbus devices in the network. Might be a collision thing haha

best regards,
Ado

ado464, 2017-06-09:
Output 1

```
mbus-serial-request-data-multi-reply -d -b 2400 /dev/ttyUSB0 2

init_slaves: debug: sending init frame #1
[2017-06-09 08:40:07] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2017-06-09 08:40:07] SEND (005): 10 40 FD 3D 16
[2017-06-09 08:40:07] SEND (005): 10 7B 02 7D 16
[2017-06-09 08:40:07] RECV (001): F6
mbus_sendrecv_request: Received invalid M-Bus response frame.
[2017-06-09 08:40:07] RECV (001): 08
[2017-06-09 08:40:08] RECV (001): 72
[2017-06-09 08:40:08] RECV (001): 30
[2017-06-09 08:40:08] RECV (001): 03
[2017-06-09 08:40:08] RECV (001): 02
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 83
[2017-06-09 08:40:08] RECV (001): 09
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 87
[2017-06-09 08:40:08] RECV (001): 03
[2017-06-09 08:40:08] RECV (001): 82
[2017-06-09 08:40:08] RECV (001): 02
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 40
[2017-06-09 08:40:08] RECV (001): 09
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): FF
[2017-06-09 08:40:08] RECV (001): 01
[2017-06-09 08:40:08] RECV (001): FF
[2017-06-09 08:40:08] RECV (001): 71
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 40
[2017-06-09 08:40:08] RECV (001): FF
[2017-06-09 08:40:08] RECV (001): 80
[2017-06-09 08:40:08] RECV (001): 82
[2017-06-09 08:40:08] RECV (001): 02
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 07
[2017-06-09 08:40:08] RECV (001): FF
[2017-06-09 08:40:08] RECV (001): 51
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 07
[2017-06-09 08:40:08] RECV (001): FF
[2017-06-09 08:40:08] RECV (001): 74
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 03
[2017-06-09 08:40:08] RECV (001): A7
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): ED
[2017-06-09 08:40:08] RECV (001): 0E
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 01
[2017-06-09 08:40:08] RECV (001): 61
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (005): 10 00 00 00 87
[2017-06-09 08:40:08] RECV (001): 00
[2017-06-09 08:40:08] RECV (001): 00
[... 1455 lines cut ...]
    </DataRecord>

    <DataRecord id="25">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>4</Tariff>
        <Device>0</Device>
        <Unit>Energy (Wh)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2017-06-09T08:42:41</Timestamp>
    </DataRecord>

    <DataRecord id="26">
        <Function>Manufacturer specific</Function>
        <Value></Value>
        <Timestamp>2017-06-09T08:42:41</Timestamp>
    </DataRecord>

</MBusData>
```

Just info if it could help out :)

best regards,
Ado

ado464, 2017-06-16:
Hello again, now a days it just returns values up to 26, and they are all random. Sometimes it reports 26 values, other 52, other times 78. Any idea what could cause this glitch?

Best regards,
Ado

ado464, 2017-06-21:
Hello, i have started getting weird output on 1 of 3 m-bus meters, it outputs to 33 values with the normal command.

Weird output:
´´
mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 1
init_slaves: debug: sending init frame #1
[2017-06-21 08:55:51] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2017-06-21 08:55:51] SEND (005): 10 40 FD 3D 16
[2017-06-21 08:55:51] SEND (005): 10 5B 01 5C 16
[2017-06-21 08:55:53] RECV (247): 68 F1 F1 68 08 01 72 15 30 50 33 A3 4C 15 02 CD 00 00 00 02 FF 34 00 00 05 FF 35 00 00 80 3F 02 FF 36 00 00 02 FF 37 00 00 02 FF 38 00 00 04 ED FF 39 00 00 01 01 05 FF 3A 00 00 C0 FF 06 FF 20 14 37 B7 00 00 00 03 FF 21 03 00 00 03 FF 22 04 00 00 03 FF 23 0B 00 00 03 FF 24 32 00 00 05 03 99 12 3F 47 05 83 FF 09 6F 12 83 3A 85 40 03 09 57 2A 47 85 40 83 FF 09 04 56 2E 41 05 83 FF 0D 99 12 3F 47 85 40 83 FF 0D 09 57 2A 47 05 83 FF 01 90 0D 7A 46 05 83 FF 02 BB 85 85 46 05 83 FF 03 B7 2F 77 46 05 FD 61 00 00 00 00 85 10 03 00 00 00 00 85 20 03 00 00 00 00 85 30 03 00 00 00 00 85 80 10 03 00 00 00 00 03 FF 25 00 00 00 05 FF 26 00 00 C8 42 03 FF 27 64 00 00 03 FF 28 03 00 00 03 FF 29 96 00 00 03 FF 2A 05 00 00 03 FF 2B 00 00 00 0F 0F 16
mbus_frame_print: Dumping M-Bus frame [type 4, 247 bytes]: 68 F1 F1 68 08 01 72 15 30 50 33 A3 4C 15 02 CD 00 00 00 02 FF 34 00 00 05 FF 35 00 00 80 3F 02 FF 36 00 00 02 FF 37 00 00 02 FF 38 00 00 04 ED FF 39 00 00 01 01 05 FF 3A 00 00 C0 FF 06 FF 20 14 37 B7 00 00 00 03 FF 21 03 00 00 03 FF 22 04 00 00 03 FF 23 0B 00 00 03 FF 24 32 00 00 05 03 99 12 3F 47 05 83 FF 09 6F 12 83 3A 85 40 03 09 57 2A 47 85 40 83 FF 09 04 56 2E 41 05 83 FF 0D 99 12 3F 47 85 40 83 FF 0D 09 57 2A 47 05 83 FF 01 90 0D 7A 46 05 83 FF 02 BB 85 85 46 05 83 FF 03 B7 2F 77 46 05 FD 61 00 00 00 00 85 10 03 00 00 00 00 85 20 03 00 00 00 00 85 30 03 00 00 00 00 85 80 10 03 00 00 00 00 03 FF 25 00 00 00 05 FF 26 00 00 C8 42 03 FF 27 64 00 00 03 FF 28 03 00 00 03 FF 29 96 00 00 03 FF 2A 05 00 00 03 FF 2B 00 00 00 0F 0F 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>33503015</Id>
        <Manufacturer>SEC</Manufacturer>
        <Version>21</Version>
        <ProductName></ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>205</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2017-06-21T08:55:53</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>1.000000</Value>
        <Timestamp>2017-06-21T08:55:53</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2017-06-21T08:55:53</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2017-06-21T08:55:53</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>0</Value>
        <Timestamp>2017-06-21T08:55:53</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
[... 459 lines cut ...]
        <Value>50.061081</Value>
        <Timestamp>2017-06-21T08:57:48</Timestamp>
    </DataRecord>

    <DataRecord id="24">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (Wh)</Unit>
        <Value>36427800</Value>
        <Timestamp>2017-06-21T08:57:48</Timestamp>
    </DataRecord>

    <DataRecord id="25">
        <Function>More records follow</Function>
        <Value></Value>
        <Timestamp>2017-06-21T08:57:48</Timestamp>
    </DataRecord>

</MBusData>
´´

Help appreciated, thank you :) 

best regards,
Ado

ado464, 2018-04-06:
Hey! Still issues with this scenario. 
I'm more specificly trying to read the Digital Input (value48) i know it's connected and should output "1" but it shows "0". Any idea what it could be?

I still need to run "mbus-serial-request-data -b 2400 /dev/ttyUSB0 2" and then 2 times 
"mbus-serial-request-data-multi-reply -b 2400 /dev/ttyUSB0 2" to output full 78 values.

It does not help only running "mbus-serial-request-data-multi-reply -b 2400 /dev/ttyUSB0 2" then it only outputs 26 values.

best regards,
Ado

lategoodbye, 2018-04-06:
@ado464 Can you provide a frame for the digital input issue where it's sure that the value must be 1?

ado464, 2018-04-09:
Hey, here is a debug output. Somehow the id 49 is value1? But i don't trust it as i don't know what it is
But if you look at the previous dumps i posted the value 49 is 0, hmm... could 49 actually be 48?

´´´
mbus-serial-request-data-multi-reply -d /dev/ttyUSB0 2
init_slaves: debug: sending init frame #1
[2018-04-09 08:11:46Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2018-04-09 08:11:46Z] SEND (005): 10 40 FD 3D 16
[2018-04-09 08:11:47Z] SEND (005): 10 7B 02 7D 16
[2018-04-09 08:11:47Z] RECV (250): 68 F4 F4 68 08 02 72 62 30 31 03 A3 4C 15 02 59 00 00 00 0D FD 0A 12 63 69 72 74 63 65 6C 45 20 72 65 64 69 65 6E 68 63 53 0D FD 0C 08 20 35 33 31 33 4D 45 69 0D FD 0E 07 37 30 30 2E 33 2E 31 03 FD 17 00 00 00 05 FD DC FF 01 10 64 09 42 05 FD DC FF 02 E2 F1 0A 42 05 FD DC FF 03 F7 44 0B 42 05 FD DC FF 00 F8 88 0A 42 05 FD C9 FF 05 41 55 C7 43 05 FD C9 FF 06 CB D9 C7 43 05 FD C9 FF 07 DB A1 C7 43 05 FD C9 FF 08 53 90 C7 43 05 FD C9 FF 01 20 29 66 43 05 FD C9 FF 02 01 2A 66 43 05 FD C9 FF 03 BF FE 66 43 05 FD C9 FF 04 A0 70 66 43 05 AE FF 01 3F 66 C2 40 05 AE FF 02 4C BC C6 40 05 AE FF 03 3B 3E C5 40 05 2E 32 98 93 41 85 40 2E 28 53 75 41 85 80 40 2E 78 E9 BF 41 05 FF 0A 09 E2 44 3F 05 FF 0B 80 1D 48 42 07 03 41 32 1B 07 00 00 00 00 1F 87 16
[2018-04-09 08:11:47Z] SEND (005): 10 5B 02 5D 16
[2018-04-09 08:11:47Z] RECV (252): 68 F6 F6 68 08 02 72 62 30 31 03 A3 4C 15 02 5A 00 00 00 07 83 FF 09 49 00 00 00 00 00 00 00 87 40 03 F6 57 6C 06 00 00 00 00 87 40 83 FF 09 00 00 00 00 00 00 00 00 04 ED FF 0C 00 00 01 01 07 83 FF 0D 41 32 1B 07 00 00 00 00 87 40 83 FF 0D F6 57 6C 06 00 00 00 00 07 83 FF 01 72 44 5D 02 00 00 00 00 07 83 FF 02 29 60 63 02 00 00 00 00 07 83 FF 03 EB 8D 5A 02 00 00 00 00 04 ED FF 0E 00 00 01 01 07 FD 61 00 00 00 00 00 00 00 00 03 FF 10 00 00 00 87 10 03 00 00 00 00 00 00 00 00 87 20 03 00 00 00 00 00 00 00 00 87 30 03 00 00 00 00 00 00 00 00 87 80 10 03 00 00 00 00 00 00 00 00 04 6D 08 09 49 24 03 FF 2C 64 00 00 03 FF 2D 00 00 00 05 FF 2E 00 00 C8 42 05 FF 2F 00 00 FA 43 03 FF 30 00 00 00 03 FD 1B 00 00 00 02 FF 32 01 00 03 FD 1A FF FF 00 1F B1 16
[2018-04-09 08:11:48Z] SEND (005): 10 7B 02 7D 16
[2018-04-09 08:11:48Z] RECV (204): 68 C6 C6 68 08 02 72 62 30 31 03 A3 4C 15 02 5B 00 00 00 02 FF 34 00 00 05 FF 35 00 00 80 3F 02 FF 36 00 00 02 FF 37 00 00 02 FF 38 00 00 04 ED FF 39 00 00 01 01 05 FF 3A 00 00 C0 FF 06 FF 20 8F 21 38 02 00 00 03 FF 21 03 00 00 03 FF 22 04 00 00 03 FF 23 0B 00 00 03 FF 24 32 00 00 05 03 6E DB E8 47 05 83 FF 09 06 81 95 3D 85 40 03 DA 79 D2 47 85 40 83 FF 09 00 00 00 00 05 83 FF 0D 6E DB E8 47 85 40 83 FF 0D DA 79 D2 47 05 83 FF 01 CE F2 1A 47 05 83 FF 02 1E 83 1C 47 05 83 FF 03 01 41 1A 47 05 FD 61 00 00 00 00 85 10 03 00 00 00 00 85 20 03 00 00 00 00 85 30 03 00 00 00 00 85 80 10 03 00 00 00 00 0F 23 16
mbus_frame_print: Dumping M-Bus frame [type 4, 250 bytes]: 68 F4 F4 68 08 02 72 62 30 31 03 A3 4C 15 02 59 00 00 00 0D FD 0A 12 63 69 72 74 63 65 6C 45 20 72 65 64 69 65 6E 68 63 53 0D FD 0C 08 20 35 33 31 33 4D 45 69 0D FD 0E 07 37 30 30 2E 33 2E 31 03 FD 17 00 00 00 05 FD DC FF 01 10 64 09 42 05 FD DC FF 02 E2 F1 0A 42 05 FD DC FF 03 F7 44 0B 42 05 FD DC FF 00 F8 88 0A 42 05 FD C9 FF 05 41 55 C7 43 05 FD C9 FF 06 CB D9 C7 43 05 FD C9 FF 07 DB A1 C7 43 05 FD C9 FF 08 53 90 C7 43 05 FD C9 FF 01 20 29 66 43 05 FD C9 FF 02 01 2A 66 43 05 FD C9 FF 03 BF FE 66 43 05 FD C9 FF 04 A0 70 66 43 05 AE FF 01 3F 66 C2 40 05 AE FF 02 4C BC C6 40 05 AE FF 03 3B 3E C5 40 05 2E 32 98 93 41 85 40 2E 28 53 75 41 85 80 40 2E 78 E9 BF 41 05 FF 0A 09 E2 44 3F 05 FF 0B 80 1D 48 42 07 03 41 32 1B 07 00 00 00 00 1F 87 16 
mbus_frame_print: Dumping M-Bus frame [type 4, 252 bytes]: 68 F6 F6 68 08 02 72 62 30 31 03 A3 4C 15 02 5A 00 00 00 07 83 FF 09 49 00 00 00 00 00 00 00 87 40 03 F6 57 6C 06 00 00 00 00 87 40 83 FF 09 00 00 00 00 00 00 00 00 04 ED FF 0C 00 00 01 01 07 83 FF 0D 41 32 1B 07 00 00 00 00 87 40 83 FF 0D F6 57 6C 06 00 00 00 00 07 83 FF 01 72 44 5D 02 00 00 00 00 07 83 FF 02 29 60 63 02 00 00 00 00 07 83 FF 03 EB 8D 5A 02 00 00 00 00 04 ED FF 0E 00 00 01 01 07 FD 61 00 00 00 00 00 00 00 00 03 FF 10 00 00 00 87 10 03 00 00 00 00 00 00 00 00 87 20 03 00 00 00 00 00 00 00 00 87 30 03 00 00 00 00 00 00 00 00 87 80 10 03 00 00 00 00 00 00 00 00 04 6D 08 09 49 24 03 FF 2C 64 00 00 03 FF 2D 00 00 00 05 FF 2E 00 00 C8 42 05 FF 2F 00 00 FA 43 03 FF 30 00 00 00 03 FD 1B 00 00 00 02 FF 32 01 00 03 FD 1A FF FF 00 1F B1 16 
mbus_frame_print: Dumping M-Bus frame [type 4, 204 bytes]: 68 C6 C6 68 08 02 72 62 30 31 03 A3 4C 15 02 5B 00 00 00 02 FF 34 00 00 05 FF 35 00 00 80 3F 02 FF 36 00 00 02 FF 37 00 00 02 FF 38 00 00 04 ED FF 39 00 00 01 01 05 FF 3A 00 00 C0 FF 06 FF 20 8F 21 38 02 00 00 03 FF 21 03 00 00 03 FF 22 04 00 00 03 FF 23 0B 00 00 03 FF 24 32 00 00 05 03 6E DB E8 47 05 83 FF 09 06 81 95 3D 85 40 03 DA 79 D2 47 85 40 83 FF 09 00 00 00 00 05 83 FF 0D 6E DB E8 47 85 40 83 FF 0D DA 79 D2 47 05 83 FF 01 CE F2 1A 47 05 83 FF 02 1E 83 1C 47 05 83 FF 03 01 41 1A 47 05 FD 61 00 00 00 00 85 10 03 00 00 00 00 85 20 03 00 00 00 00 85 30 03 00 00 00 00 85 80 10 03 00 00 00 00 0F 23 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>3313062</Id>
        <Manufacturer>SEC</Manufacturer>
        <Version>21</Version>
        <ProductName></ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>89</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

[... cut ...]
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>4</Tariff>
        <Device>0</Device>
        <Unit>Energy (Wh)</Unit>
        <Value>0.000000</Value>
        <Timestamp>2018-04-09T08:11:48Z</Timestamp>
    </DataRecord>

    <DataRecord id="78" frame="2">
        <Function>Manufacturer specific</Function>
        <Value></Value>
        <Timestamp>2018-04-09T08:11:48Z</Timestamp>
    </DataRecord>

</MBusData>
´´´

Hope this helps

best regards,
Ado

toldap, 2020-09-22:
same problem here...do you have any news?

thanks

jonaz, 2021-01-18:
Same problem here with a new Caro GNM3D-MBUS meter. Det different record count if i run it multiple times. 

Reading multi also takes a long time 5-10 seconds. Reading without multi takes milliseconds. 

I'll update this comment with some debug output tomorrow.