rscada/libmbus #126: python libmbus

cybermailer, 2017-07-26:
I want to call the bash command from python. I got a response but not that well formated as i'm doing the command directly in bash. Is this normal?

The response from python script:
`[2017-07-26 11:34:48] SEND (005): 10 40 FD 3D 16
[2017-07-26 11:34:48] SEND (005): 10 40 FD 3D 16
[2017-07-26 11:34:49] SEND (005): 10 5B 0A 65 16
[2017-07-26 11:34:49] RECV (136): 68 82 82 68 08 0A 72 10 00 00 00 2D 2C 01 04 46 00 00 00 0C 07 50 77 40 00 0C 16 01 52 08 00 0C 22 65 17 03 00 0C 59 30 27 00 00 0C 5D 91 25 00 00 0C 61 39 01 00 00 0C 2E 00 00 00 00 0C 3C 00 00 00 00 4C 07 10 71 32 00 4C 16 51 85 06 00 42 6C 1F 2C 0F 10 00 00 00 00 00 62 00 00 00 00 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 41 04 00 00 00 00 02 26 07 17 00 32 16
`
Without xml formating... 
But response directly from bash:

`init_slaves: debug: sending init frame #1
[2017-07-26 11:34:17] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2017-07-26 11:34:17] SEND (005): 10 40 FD 3D 16
[2017-07-26 11:34:17] SEND (005): 10 5B 0A 65 16
[2017-07-26 11:34:18] RECV (136): 68 82 82 68 08 0A 72 10 00 00 00 2D 2C 01 04 45 00 00 00 0C 07 50 77 40 00 0C 16 01 52 08 00 0C 22 65 17 03 00 0C 59 30 27 00 00 0C 5D 91 25 00 00 0C 61 39 01 00 00 0C 2E 00 00 00 00 0C 3C 00 00 00 00 4C 07 10 71 32 00 4C 16 51 85 06 00 42 6C 1F 2C 0F 10 00 00 00 00 00 62 00 00 00 00 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 41 04 00 00 00 00 02 26 07 17 00 31 16
mbus_frame_print: Dumping M-Bus frame [type 4, 136 bytes]: 68 82 82 68 08 0A 72 10 00 00 00 2D 2C 01 04 45 00 00 00 0C 07 50 77 40 00 0C 16 01 52 08 00 0C 22 65 17 03 00 0C 59 30 27 00 00 0C 5D 91 25 00 00 0C 61 39 01 00 00 0C 2E 00 00 00 00 0C 3C 00 00 00 00 4C 07 10 71 32 00 4C 16 51 85 06 00 42 6C 1F 2C 0F 10 00 00 00 00 00 62 00 00 00 00 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 41 04 00 00 00 00 02 26 07 17 00 31 16 
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>10</Id>
        <Manufacturer>KAM</Manufacturer>
        <Version>1</Version>
        <ProductName>Kamstrup 382 (6850-005)</ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>69</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (10 kWh)</Unit>
        <Value>407750</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume ( m^3)</Unit>
        <Value>85201</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>On time (hours)</Unit>
        <Value>31765</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (1e-2 deg C)</Unit>
        <Value>2730</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (1e-2 deg C)</Unit>
        <Value>2591</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Temperature Difference (1e-2  deg C)</Unit>
        <Value>139</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (kW)</Unit>
        <Value>0</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow (1e-2  m^3/h)</Unit>
        <Value>0</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Energy (10 kWh)</Unit>
        <Value>327110</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Volume ( m^3)</Unit>
        <Value>68551</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (date)</Unit>
        <Value>2016-12-31</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Manufacturer specific</Function>
        <Value>10 00 00 00 00 00 62 00 00 00 00 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 41 04 00 00 00 00 02 26 07 17 00</Value>
        <Timestamp>2017-07-26T11:34:18</Timestamp>
    </DataRecord>

</MBusData>
`

I'm doing:
`#!/usr/bin/python

from xml.dom import minidom
from pdb import set_trace as bkp
import subprocess

bashCommand = "mbus-serial-request-data -d -b 2400 /dev/serial0 10"
process = subprocess.Popen(bashCommand.split(), stdout=subprocess.PIPE)
output, error = process.communicate()

#mbusdata = minidom.parseString(output)
`

Maybe little thinking problem...thx for support.

kamaxeon, 2017-07-26:
Hi cibermailer,

Could you test this code ?

`
#!/usr/bin/env python
from xml.dom import minidom
import subprocess
import sys

def read_meter(bus_id):
    # Retrieve the XML data
    cmd = ['mbus-serial-request-data', '-b', '2400', ' /dev/serial0', str(bus_id)]
    reqdata = subprocess.Popen(cmd, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    mbus_xml, err = reqdata.communicate()
    if err:
        raise SystemExit, '\nERROR: "' + ' '.join(cmd) + '" returned "' + err + '"'
    try:
        return minidom.parseString(mbus_xml)
    except:
        raise SystemExit, '\nERROR: Problem parsing M-Bus XML:\n"' + mbus_xml

if __name__ == '__main__':
    result = read_meter(10)
    print(result.toxml())
`

cybermailer, 2017-07-26:
I created a file test.py on Documents folder. Set chmod a+x on this file. Runned this file by
python test.py also sudo test.py...

I'm getting this error:

`ERROR: "mbus-serial-request-data -b 2400  /dev/serial0 10" returned "mbus_serial_connect: failed to open tty.Failed to setup connection to M-bus gateway
"
`
But i can do on command line this:
`mbus-serial-request-data -b 2400  /dev/serial0 10`
without any problem...

cybermailer, 2017-07-26:
ohhh is this raspi 3 problem because of disabled access from login shell? ...i give it a try

cybermailer, 2017-07-26:
I've read a lot now -> can it be that there is some kind of problem because of:

non-ASCII characters (e.g., French or German OR some kind of timing?

ALSO:

your posts script does not make my slave led on Relay PW3 going on. Therefore there is realy no request to it.

My script makes the slave led going on and receiving data...i have both scripts checked but cannot find what can cause that error on your script...

On my script i'm now getting this:
`[2017-07-26 14:00:39] SEND (005): 10 40 FD 3D 16
[2017-07-26 14:00:40] SEND (005): 10 40 FD 3D 16
[2017-07-26 14:00:40] SEND (005): 10 5B 0A 65 16
[2017-07-26 14:00:41] RECV (136): 68 82 82 68 08 0A 72 10 00 00 00 2D 2C 01 04 58 00 00 00 0C 07 50 77 40 00 0C 16 03 52 08 00 0C 22 67 17 03 00 0C 59 82 76 00 00 0C 5D 77 49 00 00 0C 61 05 27 00 00 0C 2E 67 00 00 00 0C 3C 15 02 00 00 4C 07 10 71 32 00 4C 16 51 85 06 00 42 6C 1F 2C 0F 10 00 00 00 00 00 63 00 00 00 00 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 41 04 00 00 00 00 02 26 07 17 00 64 16
Traceback (most recent call last):
  File "test.py", line 13, in <module>
    mbusdata = minidom.parseString(output)
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 1928, in parseString
    return expatbuilder.parseString(string)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 940, in parseString
    return builder.parseString(string)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 223, in parseString
    parser.Parse(string, True)
xml.parsers.expat.ExpatError: syntax error: line 1, column 0
`

kamaxeon, 2017-07-26:
I do not think this will help, but you can try putting the utf8 encoding in the file.

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
```

Could you print the output variable ??

cybermailer, 2017-07-26:
the returning mbus frame is correct (the hex) ... on bash the hex code is the same like in python.

On bash libmbus does concert the frame automatically. With this function: mbus_frame_print: Dumping M-Bus frame [type 4, 136 bytes]

Calling mbus from python does only give the hex code. this error:

> xml.parsers.expat.ExpatError: syntax error: line 1, column 0

should mean there is something wrong, or that the hex can't be handled...some information on the ned do something like wrapping the output: <additional> + output + </additional>

i'm little confused...is there a way to transform the hex to human readable content stuff?

lategoodbye, 2017-08-16:
Yes, the test directory contains the program mbus_parse_hex, which isn't compiled per default. So you must call make within the test directory.

tnrajvignesh, 2019-01-10:
Hi,
I managed to get the libmbus library compiled and working on my pi.
I would like to use the python wrapper [here](https://github.com/rscada/python-mbus)
to develop my Mbus scanner application on Python.
The example code is [here](https://github.com/rscada/python-mbus/blob/master/examples/mbus-test-1.py)

In the example code, I see an Mbus instance being initialised in line 16
mbus = MBus(host="mbus-gw1", port=8888)

I would like to have Mbus on serial port /dev/TTYAMA0.
Please guide me on initialising Mbus on the serial port instead of TCP.