rscada/libmbus #129: The "BSD 3-clause" license should be part of the repository

magnus-lutz, 2017-08-16:
Right now the license is located at the bottom of the website http://www.rscada.se/libmbus/
It should instead be part of the repository at the project root in order to prevent legal issues as the website can potentially be modified any time.
See https://help.github.com/articles/adding-a-license-to-a-repository/ in order to learn how to easily attach a license, for instance using a "BSD 3-clause" license template

magnus-lutz, 2017-08-16:
PS: i saw the license is already included in COPYING, nevertheless its easier to have it separated as LICENSE file which also enables github to display the license directly as "BSD-3-CLAUSE" next to the "contributors" tab.
[image]

lkarsten, 2017-09-06:
Further on, most source files in mbus/ contain this header:
```
// Copyright (C) 2010-2011, Robert Johansson, Raditex AB
// All rights reserved.
```

which I believe is not consistent with a 3 clause BSD license.

See for example https://github.com/varnishcache/varnish-cache/blob/master/bin/varnishd/cache/cache_ban.c for how a 2-clause bsd project annotate source files.