rscada/libmbus #135: M-Bus tcp transport layer response timeout has been reached

goeh, 2018-03-28:
We are using libmbus to read data from an Elster H5000 water meter via CMe3000 tcp gateway. But most request fails after a few seconds with an error. Only 1/3 of requests succeeds.

"Failed to receive M-Bus response frame: M-Bus tcp transport layer response timeout has been reached."

I located the error message to https://github.com/rscada/libmbus/blob/3d5b865ebe359f5de5aabf1cc7ab454e187319af/mbus/mbus-tcp.c#L212 but I don't have a clue why it's happening. Since it works occasionally my guess is that it'a a timeout problem. What do you think? Any help appreciated.

https://www.elstermetering.com/en/product-detail/470/en/H5000
https://www.elvaco.se/en/product/infrastructure1/cme3000-m-bus-gateway-for-fixed-network--1050015

Apollon77, 2018-03-28:
The default Timeout for TCP are 4 seconds ... you can try to use "mbus_tcp_set_timeout_set(timeout);" (timeout in seconds as double value) to increase the timeout, but you need to call it before mbus_connect

Can you try that please to see if with higher timrouts your device is working better?

goeh, 2018-03-28:
Thanks @Apollon77 , that was really helpful. I changed the timeout to 8 seconds and I can confirm that it now took 8 seconds to get the error message. But now I suspect the M-Bus/TCP gateway (CMe3000) because after 5 failed requests I rebooted the CMe3000 and then the first request worked ok immediately. After 6 successful requests it failed again. There is something strange with that gateway...

Apollon77, 2018-03-28:
Yes seems so, so no issue for libmbus :-)

goeh, 2018-03-28:
I'll close this issue in next couple of days when we've finished debugging the problem.

Apollon77, 2018-04-04:
For your info, maybe it helps: I also had a comparable problem with an other user. It seems that his TCP gateway (other then yours) closes the connection after some time. We ended up in always doing a connection make the needed queries, the close and reopen connection directly before next query. And we introduced a minimum delay betweek two connections. But maybe something comparable.

goeh, 2018-04-05:
Our usecase opens a new connection to the gateway every 30 minutes and reads data. The connection is closed after each reading.
I added a retry-loop around the request that waits 15 seconds between retries. Not a nice solution but I hope it will work better this way.