rscada/libmbus #138: Cannot Find MBus meter using Libmbus: Collisions at all 250 addresses

MZourob, 2018-05-09:
I am trying to connect to an MBus meter and using a TechBase MBus 10 converter to RS232. Then I use an RS232 to USB and connected to a RPI3 with Raspian. Every time I scan (mbus-serial-scan -b 2400 /dev/ttyUSB0), it reports (Collision at address ###) for the addresses from 0 to 250. And it does not detect the meter. I have tried that with two meters with no luck. Can anyone give me any insights to this issue?

Thanks alot.

Apollon77, 2018-05-09:
Can you compile with debug and post the output?! Or at all the complete output?

lategoodbye, 2018-05-10:
Please provide the output after adding the debug parameter "-d".
Did you try it with a PC?

MZourob, 2018-05-10:
Thanks a lot for the replies. This is a sample of the output I am getting:

Scanning primary addresses:
0 [2018-05-10 13:17:15] SEND (005): 10 40 00 40 16
[2018-05-10 13:17:15] RECV (001): 00

[2018-05-10 13:17:15] RECV (001): 00
[2018-05-10 13:17:15] RECV (001): 00
[2018-05-10 13:17:15] RECV (001): 40
[2018-05-10 13:17:15] RECV (001): 04
Collision at address 0
1 [2018-05-10 13:17:16] SEND (005): 10 40 01 41 16
[2018-05-10 13:17:16] RECV (001): 00

[2018-05-10 13:17:16] RECV (001): 00
[2018-05-10 13:17:16] RECV (001): 00
[2018-05-10 13:17:16] RECV (001): 00
[2018-05-10 13:17:16] RECV (001): 04
Collision at address 1
2 [2018-05-10 13:17:16] SEND (005): 10 40 02 42 16
[2018-05-10 13:17:16] RECV (001): 00

[2018-05-10 13:17:16] RECV (001): 00
[2018-05-10 13:17:16] RECV (001): 00
[2018-05-10 13:17:16] RECV (001): 02
[2018-05-10 13:17:16] RECV (001): 04
Collision at address 2
3 [2018-05-10 13:17:16] SEND (005): 10 40 03 43 16
[2018-05-10 13:17:16] RECV (001): 00

[2018-05-10 13:17:16] RECV (001): 00
[2018-05-10 13:17:16] RECV (001): 03
[2018-05-10 13:17:16] RECV (001): 02
[2018-05-10 13:17:16] RECV (001): 04
Collision at address 3
4 [2018-05-10 13:17:16] SEND (005): 10 40 04 44 16
[2018-05-10 13:17:16] RECV (005): 10 00 00 00 06

Collision at address 4
5 [2018-05-10 13:17:16] SEND (005): 10 40 05 45 16
[2018-05-10 13:17:17] RECV (001): 00

[2018-05-10 13:17:17] RECV (001): 00
[2018-05-10 13:17:17] RECV (001): 00
[2018-05-10 13:17:17] RECV (001): 00
[2018-05-10 13:17:17] RECV (001): 04
Collision at address 5
6 [2018-05-10 13:17:17] SEND (005): 10 40 06 46 16
[2018-05-10 13:17:17] RECV (001): 00
 

And it goes on scanning all 250 addresses and have collisions at all of them.

MZourob, 2018-05-10:
If I set the baud rate to 300, This is the output:

0 [2018-05-10 13:30:43] SEND (005): 10 40 00 40 16
[2018-05-10 13:30:43] RECV (005): 10 40 00 40 16

1 [2018-05-10 13:30:43] SEND (005): 10 40 01 41 16
[2018-05-10 13:30:43] RECV (005): 10 40 01 41 16

2 [2018-05-10 13:30:43] SEND (005): 10 40 02 42 16
[2018-05-10 13:30:43] RECV (005): 10 40 02 42 16

3 [2018-05-10 13:30:43] SEND (005): 10 40 03 43 16
[2018-05-10 13:30:44] RECV (005): 10 40 03 43 16

4 [2018-05-10 13:30:44] SEND (005): 10 40 04 44 16
[2018-05-10 13:30:44] RECV (005): 10 40 04 44 16

5 [2018-05-10 13:30:44] SEND (005): 10 40 05 45 16
[2018-05-10 13:30:44] RECV (005): 10 40 05 45 16

6 [2018-05-10 13:30:44] SEND (005): 10 40 06 46 16
[2018-05-10 13:30:44] RECV (005): 10 40 06 46 16

7 [2018-05-10 13:30:44] SEND (005): 10 40 07 47 16
[2018-05-10 13:30:44] RECV (005): 10 40 07 47 16

8 [2018-05-10 13:30:44] SEND (005): 10 40 08 48 16
[2018-05-10 13:30:45] RECV (005): 10 40 08 48 16

9 [2018-05-10 13:30:45] SEND (005): 10 40 09 49 16
[2018-05-10 13:30:45] RECV (005): 10 40 09 49 16

10 [2018-05-10 13:30:45] SEND (005): 10 40 0A 4A 16
[2018-05-10 13:30:45] RECV (005): 10 40 0A 4A 16

11 [2018-05-10 13:30:45] SEND (005): 10 40 0B 4B 16
[2018-05-10 13:30:45] RECV (005): 10 40 0B 4B 16

12 [2018-05-10 13:30:45] SEND (005): 10 40 0C 4C 16
[2018-05-10 13:30:45] RECV (005): 10 40 0C 4C 16

13 [2018-05-10 13:30:45] SEND (005): 10 40 0D 4D 16
[2018-05-10 13:30:46] RECV (005): 10 40 0D 4D 16

14 [2018-05-10 13:30:46] SEND (005): 10 40 0E 4E 16
[2018-05-10 13:30:46] RECV (005): 10 40 0E 4E 16

...

lategoodbye, 2018-05-10:
So 300 baud works except the fact that the Meter doesn't answer because of wrong baudrate?
Please provide more information:
- Which libmbus version do you use?
- Which Raspberry Pi kernel / OS do you use?

I guess you have a local echo. Back in 2012 implemented a simple echo cancelation, but this isn't perfect. It's the best to fix this in hardware or configuration.

MZourob, 2018-05-10:
I am using the latest one online (http://www.rscada.se/libmbus/). libmbus-0.8.0.tar.gz. 
This is the version returned  by uname -a: Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.3 (stretch)
Release:        9.3
Codename:       stretch

lategoodbye, 2018-05-10:
Unfortunately the "latest" release was in 2012. You better take the current state from github. This wont fix your problem. Your Raspberry Pi is receiving garbage and libmbus doesn't any chance to decode it.

Here my ideas to narrow down the issue:
- Connect the Raspberry Pi via USB to RS232 adapter to a PC and check the UART communication
- Use a M-Bus Master with Echo Cancelation
- Try the setup on a PC instead of Raspberry Pi
- Make sure that your USB to RS232 adapter provides the necessary power on the handshake lines

Apollon77, 2018-05-10:
I also did a check on Windows with mbus readout Software first and after that worked I had all Infos also to use on Linux.

MZourob, 2018-05-11:
I am using an FTDI USB/RS232 convertor (https://www.digikey.ca/products/en/computers-office-components-accessories/adapters-converters/882?k=rs232%20to%20UARt). The usb is connected to the PI and the RS232 side is connected to the Master. 

I will try it with windows and report the results back

MZourob, 2018-05-11:
I am not sure if I am testing it properly on windows, but I have installed bash and followed the same procedure. Now after installing everything from rscada, it gives the following error everytime I try to scan

mbus_serial_connect: failed to open tty.mbus_connect_serial: Failed to setup serial connection to M-bus gateway on /dev/ttyUSB0.
Failed to setup connection to M-bus gateway

Apollon77, 2018-05-11:
On windows you need to use specific MBus software ... google for it ... There are also free versions.
I used https://www.relay.de/en/products/software/mbconf/

rscada library is not working under windows ... I have an experimental windows-build-branch in my fork but without all the binaries at the moment.

lategoodbye, 2018-05-14:
Did you tried the combination PC + Linux + MBus master + USB adapter?

MZourob, 2018-05-14:
I would just like to mention that I a bit new to the whole Linux thing but I am figuring out through reading and searching online ... just in case I asked some trivial question :) 

If you mean running Linux on a Windows maching, then yes I have tried that as well and it gave me an error. I will replicate the error and insert it in the comment now.

MZourob, 2018-05-14:
There is also this flag everytime I run the ./configure command

Compiler flags:         -g -O2

MZourob, 2018-05-14:
No, from Windows 10 store, I have downloaded an Ubuntu (bash?) and used it. I dont know if that counts as a virtual machine. Here is a screenshot.

[image]

MZourob, 2018-05-14:
It seems I misunderstood what you meant by "PC + Linux + MBus Master + USB Adapter". Could you please elaborate on that?

lategoodbye, 2018-05-14:
Sorry, this won't work. You will need a native Linux (for example Ubuntu, Debian) running on your PC.

MZourob, 2018-05-14:
I have just installed xubuntu on my PC and still the same error. As if the adapter is not working with Linux on PC for some reason. Although it did work with the Pi

lategoodbye, 2018-05-15:
It's possible that the serial adapter introduce a too long delay. I will prepare a test branch.

MZourob, 2018-05-15:
@lategoodbye Hi Stephan, 

Thank you very much for your followups. I have figured out the problem and it also relates to issue #84. I was installing and building the package in the normal user mode while using sudo. However, i tried everything in the (su) mode as root. And it all worked perfectly. Now the question is: what are the packages that are involved in the library or installation that needs an (su) permission level? 

When I connect three meters, and scan through the primary addresses, it gives a collision at address 0. Just to confirm with you, that has to do with the primary address on the actual meters, is not it? This means that all if the meters are set to a primary address of 0.

When I tried scanning through the secondary addresses, it showed all three different meters and I was able to pull data from them using the mbus-serial-request-data command and their individual addresses.

Just to confirm, when we connect the meters, they are all connected in parallel, is not it?

lategoodbye, 2018-05-15:
This confuses me. What did you change to make it working? Does is works under Raspberry Pi, too?

Answers to your questions:

1. libmbus doesn't require special permission, but  access to serial interfaces like /dev/ttyUSB0 does.
2. Yes, primary addresses must be configured before using. Otherwise you will get collisions.
3. In case you have general questions about M-Bus please look at http://www.m-bus.com/files/MBDOC48.PDF

lategoodbye, 2018-05-15:
Nevertheless i prepared a extended version of mbus-serial-scan, which provides an option to specify an additional timeout in 1/10 seconds. Btw i reverted my last change to the serial timeouts.

Example with 1 second additional delay:
mbus-serial-scan -d -t 10 /dev/ttyUSB0

Relevant branch:
https://github.com/rscada/libmbus/tree/adjustable-timeout

@Apollon77 What's your opinion?

MZourob, 2018-05-15:
@lategoodbye literally all what I did is install the last library while being in su mode and it worked perfectly.

At first I installed the libmbus on the third party Pi, which had its own MBus code and always launches in su mode. The libmbus worked perfectly. This gave me the idea to try the libmbus in su mode on my RPi3 and that’s what I have done. Literally did not change a thing in the setup, and it worked perfectly on both my RPi and the Linux on PC as well.

lategoodbye, 2018-05-15:
Usually you also always need superuser rights to install libraries. libmbus doesn't have any influence on that. After installation it's necessary to call `ldconfig`, which also requires higher privileges.

Apollon77, 2018-05-15:
@lategoodbye change seems reasonable. As soon as you officially commit that I can take the new default timings in my node-lib and receive user feedback too .. and if needed can build the same logic to use the new offset for slower systems if users need that.

MZourob, 2018-05-18:
@lategoodbye Hello,

I cannot find the mbus-serial-set-address command in the libmbus. Is it by any chance included in an updated version of the library?

lategoodbye, 2018-05-18:
I don't have access to the rscada website. How about downloading the latest version from github?

MZourob, 2018-05-18:
That works nicely. Thank you very much Stefan.

lategoodbye, 2018-05-19:
stefan.wahren@i2se.com

MZourob, 2018-05-30:
Hi Stefan @lategoodbye,

Thank you for all the help so far. I was wondering, is there a way to read stored data in the meter? The meter at hand stores up to 30 days of data. So I was wondering if I can access that data using the Libmbus library. 

Thank you.

MZourob, 2018-06-08:
Hi Stefan @lategoodbye 

Hope this message finds you well. 

I was wondering if the library can be used to retrieve historical data from the meters. As of now, it pulls the instantaneous values from the meter, However, what about historical data that is stored in the meter?

lategoodbye, 2018-06-17:
As long as the datapoints are described within the standard or MBDOC48 it should be possible. Sorry, how should i answer such a unspecific question.

MZourob, 2018-06-27:
Hi @lategoodbye,

I have figured out that this issue would have to do with the meter firmware and not with the library. I was able to retrieve historical data from the meter that was originally stored by the meter as past variables. So I get back from the meter say 20 variables with 7 of them historical values.

Thanks for the feedback.

nardev, 2018-06-29:
@MZourob since you are new to this, i have couple of simple questions but i would rather avoid asking it here. Could you send me an email?

MZourob, 2018-06-29:
Mohammed.zourob@sagacitysoft.ca<mailto:Mohammed.zourob@sagacitysoft.ca>

Best Regards,
Mohammed Omar Zourob, Ph.D., M.E.Sc.
Electrical/Electronics Engineer