rscada/libmbus #139: Just help, how to start?

nardev, 2018-06-29:
Sorry for opening an issue for this.

I just found out about m-bus and if somebody is happy to help, i have following concerns:

1. In my building, just next to my doors, in a locker, is main water meter. I realized it's M-Bus Cyble v2.0 (https://www1.itron.com/local/Poland%20Product%20Portolio/GA-CYBLEMBUSSTD-01-EN-10-14.pdf)
It's one of 4 in the locker and all 4 are connected to main line with two wires.

My concern is, is it safe to hook RPI/ESP8266 directly to the main line? (i don't want to damage anything)
Would i be able to identify my meter and practically, how to do that?

2. Did anyone try to rebuild this lib for Arduino and use it on ESP8266?

3. Since i have no idea about the protocol and the device, does the "reading" device have to be on always in order to catch all impulses or the given value is the last value on the device, so occasional reading is just enough for status update?

4. Since it's quite impractical for me to stay on the lobby and play with the devices, what would the easiest way to test if i can read data and what i get once i read it.

Thank you all

lategoodbye, 2018-06-30:
0. Please look at media converters at https://github.com/rscada/libmbus/wiki/Supported-devices
1. & 3. Please read http://www.m-bus.com/files/MBDOC48.PDF
2. Take a look at https://github.com/rscada/libmbus/issues/27
4. I recommend to use your own devices for tests.

nardev, 2018-06-30:
Good thing, Itron is there on the list, but my device is cyble 2.0. I hope that would be ok.
[image]

Regarding arduino, i found this project, however, very little hardware tips:

[MBusMqttLogger](https://github.com/roarfred/MBusMqttLogger)

and the derived one:
[wifi-mbus](https://github.com/emard/wifi-mbus)

lategoodbye, 2018-06-30:
I think you misunderstand me. You will need a M-Bus master, too (please refer to media converter as mentioned in the link above). The cyble 2.0 should work, but i never test it,

nardev, 2018-06-30:
Can there be two masters on the line?

I assume, they already have some master there on the line, i just want to get data from my meter.

Actually, it would be great to get simplified explanation how the m-bus network works, what is where...

closed 2019-02-22 by lategoodbye