rscada/libmbus #141: LandisGyr G350

yctn, 2018-10-13:
Hello, im trying to read out my gas meter.
it seems that someone already done it.
https://github.com/rscada/libmbus/issues/110
i have a laptop with debian 9 installed.
im using this as a IR head on the optical port on the gas meter. http://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-usb-ausgang

when i run mbus-serial-scan -b 300 /dev/ttyUSB0 i does not find any addresses so i dont know how i should proceed from here.

hope someone can advice me on this.

yctn, 2018-10-15:
@Apollon77 directly cloned from github
@lategoodbye hmm.. well it seems https://github.com/rscada/libmbus/issues/110 has done it. but i dont know how he connected it all upcause he uses the same device as i am using so it should be possible im only dont know how exacly

lategoodbye, 2018-10-15:
Not every meter of this type support M-Bus. You will need a suitable M-Bus communication module for this meter.

yctn, 2018-10-15:
@lategoodbye the meter has a optical port on it so i plugged this on it:http://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-usb-ausgang

Apollon77, 2018-10-15:
And it is really mbus protocol? Normally with IR you have D0 or such protocols

yctn, 2018-10-15:
@Apollon77 to be honest i dont know but on another forum they told me it mbus 
https://www.gurux.fi/node/12944

Apollon77, 2018-10-15:
Accordingto the datasheets i found wired mbus or wireless mbus is possible ... but both not using ir! I was not able to find Infos to ir at all. Do you have the exact datasheet for your device?

yctn, 2018-10-17:
hmm so then i would need to buy a wireless mbus usb adapter?

this data sheets i have found
https://www.landisgyr.eu/webfoo/wp-content/uploads/2012/09/Libra_350_i_data_sheet12.pdf
https://www.landisgyr.eu/product/landisgyr-libra-350i-series/

lategoodbye, 2018-10-17:
libmbus hasn't been tested with wireless mbus. I assume it doesn't work.

Since we didn't sell you the meter, please contact your distributor which communication module is installed.

Apollon77, 2018-10-17:
wireles mbus is different, so not directly compatible

closed 2019-02-22 by lategoodbye