rscada/libmbus #145: Sharky 775 Diehl

josvandekoffie, 2019-02-19:
Hello great programmer,
I use Mbus for the Sharky but somehow sometimes I have a 1500 in front of power and differential temperature number. This happens especially when the power read out is close to zero I think because the differential temperature and poweroutput have the problem at the same moments in time.
In the graphs on Domoticz this shows as huge spikes. Curious if someone can solve this puzzle. Maybe the 15-- part needs to be a minus sign or so?

**System:**
Mbus master (usb version AliXpress)
Sharky 775
Raspberry Pi 3

**Installation:**
https://github.com/rscada/libmbus/archive/master.zip

**Manual of Sharky 775:**
https://www.wms.nl/application/files/4615/0607/4805/Sharky775_M-Bus_telegram_E.pdf

pi@raspberrypi:/usr/local/bin $ mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 61
init_slaves: debug: sending init frame #1
[2019-01-30 22:20:26Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2019-01-30 22:20:27Z] SEND (005): 10 40 FD 3D 16
[2019-01-30 22:20:27Z] SEND (005): 10 5B 3D 98 16
[2019-01-30 22:20:28Z] RECV (110): 
68 68 68 68 08 3D 72 
00 38 71 41 
24 23 2F 04 C2 00 00 00 0C 06 31 40 00 00 8C 10 06 00 00 00 00 8C 20 13 20 43 16 00 0C 13 72 90 33 01 0C 2B 87 11 00 F0 0B 3B 26 08 00 0A 5A 27 03 0A 5E 42 03 0A 62 15 F0 0A 27 54 26 04 6D 02 02 7F 21 4C 06 00 00 00 00 4C 13 00 00 00 00 CC 10 06 00 00 00 00 CC 20 13 00 00 00 00 42 6C 5F 25 F2 16
mbus_frame_print: Dumping M-Bus frame [type 4, 110 bytes]: 68 68 68 68 08 3D 72 00 38 71 41 24 23 2F 04 C2 00 00 00 0C 06 31 40 00 00 8C 10 06 00 00 00 00 8C 20 13 20 43 16 00 0C 13 72 90 33 01 0C 2B 87 11 00 F0 0B 3B 26 08 00 0A 5A 27 03 0A 5E 42 03 0A 62 15 F0 0A 27 54 26 04 6D 02 02 7F 21 4C 06 00 00 00 00 4C 13 00 00 00 00 CC 10 06 00 00 00 00 CC 20 13 00 00 00 00 42 6C 5F 25 F2 16
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>41713800</Id>
        <Manufacturer>HYD</Manufacturer>
        <Version>47</Version>
        <ProductName></ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>194</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>4031</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>1</Tariff>
        <Device>0</Device>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>2</Tariff>
        <Device>0</Device>
        <Unit>Volume (m m^3)</Unit>
        <Value>164320</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>1339072</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (W)</Unit>
        <Value>150001187</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow (m m^3/h)</Unit>
        <Value>826</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (1e-1 deg C)</Unit>
        <Value>327</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (1e-1 deg C)</Unit>
        <Value>342</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Temperature Difference (1e-1  deg C)</Unit>
        <Value>15015</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Operating time (days)</Unit>
        <Value>2654</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2019-01-31T02:02:00</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="12">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>0</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="13">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Tariff>1</Tariff>
        <Device>0</Device>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="14">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Tariff>2</Tariff>
        <Device>0</Device>
        <Unit>Volume (m m^3)</Unit>
        <Value>0</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

    <DataRecord id="15">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (date)</Unit>
        <Value>2018-05-31</Value>
        <Timestamp>2019-01-30T22:20:28Z</Timestamp>
    </DataRecord>

</MBusData>

lategoodbye, 2019-02-19:
Okay, could you please provide a hexdump of a good case telegram?

josvandekoffie, 2019-02-20:
Here is one just freshly baked. If I perform the data request twice I directly get an error. See at the bottom. Don't know if this has anything to do with it though.

pi@raspberrypi:~/domoticz/scripts $ sudo /usr/local/bin/mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 61
init_slaves: debug: sending init frame #1
[2019-02-20 13:09:56Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2019-02-20 13:09:56Z] SEND (005): 10 40 FD 3D 16
[2019-02-20 13:09:57Z] SEND (005): 10 5B 3D 98 16
[2019-02-20 13:09:57Z] RECV (110): 68 68 68 68 08 3D 72 00 38 71 41 24 23 2F 04 C2 00 00 00 0C 06 43 50 00 00 8C 10 06 00 00 00 00 8C 20 13 53 81 17 00 0C 13 54 06 65 01 0C 2B 67 00 00 00 0B 3B 13 06 00 0A 5A 14 02 0A 5E 13 02 0A 62 01 00 0A 27 75 26 04 6D 34 10 74 22 4C 06 00 00 00 00 4C 13 00 00 00 00 CC 10 06 00 00 00 00 CC 20 13 00 00 00 00 42 6C 5F 25 E9 16
mbus_frame_print: Dumping M-Bus frame [type 4, 110 bytes]: 68 68 68 68 08 3D 72 00 38 71 41 24 23 2F 04 C2 00 00 00 0C 06 43 50 00 00 8C 10 06 00 00 00 00 8C 20 13 53 81 17 00 0C 13 54 06 65 01 0C 2B 67 00 00 00 0B 3B 13 06 00 0A 5A 14 02 0A 5E 13 02 0A 62 01 00 0A 27 75 26 04 6D 34 10 74 22 4C 06 00 00 00 00 4C 13 00 00 00 00 CC 10 06 00 00 00 00 CC 20 13 00 00 00 00 42 6C 5F 25 E9 16
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>41713800</Id>
        <Manufacturer>HYD</Manufacturer>
        <Version>47</Version>
        <ProductName></ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>194</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>5043</Value>
        <Timestamp>2019-02-20T13:09:57Z</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>1</Tariff>
        <Device>0</Device>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2019-02-20T13:09:57Z</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>2</Tariff>
        <Device>0</Device>
        <Unit>Volume (m m^3)</Unit>
        <Value>178153</Value>
        <Timestamp>2019-02-20T13:09:57Z</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>1650654</Value>
        <Timestamp>2019-02-20T13:09:57Z</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (W)</Unit>
        <Value>67</Value>
        <Timestamp>2019-02-20T13:09:57Z</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
[... 81 lines cut ...]

    <DataRecord id="15">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (date)</Unit>
        <Value>2018-05-31</Value>
        <Timestamp>2019-02-20T13:09:57Z</Timestamp>
    </DataRecord>

</MBusData>
pi@raspberrypi:~/domoticz/scripts $ sudo /usr/local/bin/mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 61
init_slaves: debug: sending init frame #1
[2019-02-20 13:13:47Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2019-02-20 13:13:47Z] SEND (005): 10 40 FD 3D 16
[2019-02-20 13:13:47Z] SEND (005): 10 5B 3D 98 16
Failed to receive M-Bus response frame.
pi@raspberrypi:~/domoticz/scripts $ sudo /usr/local/bin/mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 61
init_slaves: debug: sending init frame #1
[2019-02-20 13:13:48Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2019-02-20 13:13:49Z] SEND (005): 10 40 FD 3D 16
[2019-02-20 13:13:49Z] SEND (005): 10 5B 3D 98 16
Failed to receive M-Bus response frame.
pi@raspberrypi:~/domoticz/scripts $ sudo /usr/local/bin/mbus-serial-request-data -d -b 2400 /dev/ttyUSB0 61

lategoodbye, 2019-02-20:
@josvandekoffie Could you please test branch negative-bcd?

josvandekoffie, 2019-02-20:
Hi, just installed the libmbus-negative-bcd.zip. I will let it run for a day to see how it goes!

josvandekoffie, 2019-02-21:
Good evening! Just wanted to let you know: It works perfectly with the negative-bcd branch! Thank you so much!

lategoodbye, 2019-02-22:
I've merged the branch into master.

closed 2019-02-22 by lategoodbye