rscada/libmbus #147: 10^9 is Giga not Terra

lvogt, 2019-06-28:
https://github.com/rscada/libmbus/blob/2dc3daaf7080cb77640e6caee3154585b8196a37/mbus/mbus-protocol.c#L1826

10^9 is Giga.

On a side note: I think outputting the data with this mix of metric prefixes and "number multipliers" is not really a good idea...

Apollon77, 2019-07-05:
@lvogt is this an issue in ioBroker? Then I can fix it in my fork for node-mbus and also deliver as PR for here .. ?!

lvogt, 2019-07-08:
My parsing code knows of this bug and treats T as 10^9...
However, I think this is more of a theoretical problem as I don't think there are many meters which output values in this range...

Apollon77, 2019-07-08:
Ok, then I wait for an official merge or repsonse and we update later

lategoodbye, 2019-07-08:
Sorry, i'm busy with BCM2835 stuff. But any PR is welcome.

closed 2019-07-08 by lategoodbye