rscada/libmbus #149: How to send different command

prein2, 2019-09-29:
The mbus-serial-request-data and mbus-serial-request-data-multi-reply commands always first send the init message: 10 40 FD 3D 16
Next message is the 10 7B 01 7C 16 command to read the data.

My Aestron energy meter is reporting Energy when this message is sent. When I want to read Voltage, Current and power, I have to send another message: 68 03 03 68 53 01 b1 05 16
 
I'm failing to figure out how to send this message. Anyone who can help?

BTW, M-bus protocol of Aestron 220 and 630 meter can be found here:
https://www.kwhmeter.nl/Files/2/26000/26485/Attachments/Product/0Oq90EEQ42226h7911o5983365exE52r.pdf

beireken, 2019-10-03:
I have the same issue using these meters!

prein2, 2019-10-03:
I managed to do this with the pymeterbus library on github. It is possible with libmbus, but I am not that good at C programming. With python programming I fixed it, so using pymeterbus right now..