rscada/libmbus #163: Running the tests on unmodified master gives printouts (diffs)

fredrik-sk, 2020-06-05:
I created a Dockerfile to run the tests, and ran the tests on an unmodified master.
And I got a bunch of printouts (diff).

https://github.com/fredrik-sk/libmbus/runs/742450655?check_suite_focus=true

fredrik-sk, 2020-06-08:
Update: Some more research showed that this was introduced by commit 2f9fa5ccc8aa032556ab77b7cafcb24670b6bf8d, the code was updated but not the tests.
Reverting that commit makes the test run clean again.

I will look into this some more.

Added a comment:
According to wikipedia ( https://en.wikipedia.org/wiki/Binary-coded_decimal , section Packed BCD), 0xF0 should be interpreted as a positve number.

https://github.com/rscada/libmbus/commit/2f9fa5ccc8aa032556ab77b7cafcb24670b6bf8d#r39755040

lategoodbye, 2020-06-27:
I think, i've found a solution. Unfortunately we have a regression in #151 which must be fixed first.

Apollon77, 2020-06-30:
I also have such a case ... I brought my branch of libmbus which is used to buil "node-mbus" to the current status of the master but my Node-mbs tests fail because also here I have one case where the "SlaveInformation.Id" field changes.

I already have the "improved BCD decoding" commit in too!

lategoodbye, 2020-07-01:
@Apollon77 Your hex contains 3E020005 which is actually slave id 500023E. I'm not sure it's allowed to have hex digits in a slave id and i don't know this comes from a real device. But make libmbus hiding this issue to make your tests succeed doesn't make sense to me.

So please describe your expectations.

Apollon77, 2020-07-01:
@lategoodbye if it is that way I'm fine ... then this change is "simply because decoding was broken before" and so it is fixed now :-) I was just wondering which is right ... When I remember correctly I had that "data string" from a real device. Then I fix the test expectation!

lategoodbye, 2020-07-08:
I consider this as fixed

closed 2020-07-08 by lategoodbye