rscada/libmbus #174: Debian build seems to be broken?

sjlongland, 2020-07-14:
We're trying to build a binary `.deb` package for `libmbus` so we can distribute that instead of statically linking it.  It would appear this build references files that no longer exist:

```
14-Jul-2020 16:07:00 | make[3]: Leaving directory '/mnt/src/obj-x86_64-linux-gnu'
-- | --
14-Jul-2020 16:07:00 | make -f CMakeFiles/libmbus.dir/build.make CMakeFiles/libmbus.dir/build
14-Jul-2020 16:07:00 | make[3]: Entering directory '/mnt/src/obj-x86_64-linux-gnu'
14-Jul-2020 16:07:00 | make[3]: Nothing to be done for 'CMakeFiles/libmbus.dir/build'.
14-Jul-2020 16:07:00 | make[3]: Leaving directory '/mnt/src/obj-x86_64-linux-gnu'
14-Jul-2020 16:07:00 | [100%] Built target libmbus
14-Jul-2020 16:07:00 | make[2]: Leaving directory '/mnt/src/obj-x86_64-linux-gnu'
14-Jul-2020 16:07:00 | /usr/bin/cmake -E cmake_progress_start /mnt/src/obj-x86_64-linux-gnu/CMakeFiles 0
14-Jul-2020 16:07:00 | make -f CMakeFiles/Makefile2 preinstall
14-Jul-2020 16:07:00 | make[2]: Entering directory '/mnt/src/obj-x86_64-linux-gnu'
14-Jul-2020 16:07:00 | make[2]: Nothing to be done for 'preinstall'.
14-Jul-2020 16:07:00 | make[2]: Leaving directory '/mnt/src/obj-x86_64-linux-gnu'
14-Jul-2020 16:07:00 | Install the project...
14-Jul-2020 16:07:00 | /usr/bin/cmake -P cmake_install.cmake
14-Jul-2020 16:07:00 | -- Install configuration: "None"
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/share/pkgconfig/libmbus.pc
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/lib/x86_64-linux-gnu/libmbus.so
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/lib/x86_64-linux-gnu/cmake/libmbus/libmbusTargets.cmake
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/lib/x86_64-linux-gnu/cmake/libmbus/libmbusTargets-none.cmake
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/lib/x86_64-linux-gnu/cmake/libmbus/libmbusConfig.cmake
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/lib/x86_64-linux-gnu/cmake/libmbus/libmbusConfigVersion.cmake
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/include/mbus
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/include/mbus/mbus-protocol-aux.h
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/include/mbus/mbus-protocol.h
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/include/mbus/mbus-serial.h
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/include/mbus/mbus-tcp.h
14-Jul-2020 16:07:00 | -- Installing: /mnt/src/debian/tmp/usr/include/mbus/mbus.h
14-Jul-2020 16:07:00 | make[1]: Leaving directory '/mnt/src/obj-x86_64-linux-gnu'
14-Jul-2020 16:07:00 | dh_install
14-Jul-2020 16:07:00 | dh_install: Compatibility levels before 9 are deprecated (level 7 in use)
14-Jul-2020 16:07:00 | dh_install: Cannot find (any matches for) "usr/lib/lib*.a" (tried in "." and "debian/tmp")
14-Jul-2020 16:07:00 | dh_install: libmbus-dev missing files: usr/lib/lib*.a
14-Jul-2020 16:07:00 | dh_install: Cannot find (any matches for) "usr/lib/lib*.so" (tried in "." and "debian/tmp")
14-Jul-2020 16:07:00 | dh_install: libmbus-dev missing files: usr/lib/lib*.so
14-Jul-2020 16:07:00 | dh_install: Cannot find (any matches for) "usr/lib/*.la" (tried in "." and "debian/tmp")
14-Jul-2020 16:07:00 | dh_install: libmbus-dev missing files: usr/lib/*.la
14-Jul-2020 16:07:00 | dh_install: Cannot find (any matches for) "usr/lib/pkgconfig/libmbus.pc" (tried in "." and "debian/tmp")
14-Jul-2020 16:07:00 | dh_install: libmbus-dev missing files: usr/lib/pkgconfig/libmbus.pc
14-Jul-2020 16:07:00 | dh_install: Cannot find (any matches for) "usr/lib/lib*.so.*" (tried in "." and "debian/tmp")
14-Jul-2020 16:07:00 | dh_install: libmbus1 missing files: usr/lib/lib*.so.*
14-Jul-2020 16:07:00 | dh_install: Cannot find (any matches for) "usr/bin/*" (tried in "." and "debian/tmp")
14-Jul-2020 16:07:00 | dh_install: libmbus1 missing files: usr/bin/*
14-Jul-2020 16:07:00 | dh_install: missing files, aborting
14-Jul-2020 16:07:00 | make: *** [binary] Error 2
14-Jul-2020 16:07:00 | debian/rules:6: recipe for target 'binary' failed
14-Jul-2020 16:07:00 | dpkg-buildpackage: error: debian/rules binary gave error exit status 2
```

Culprit seems to be this line: https://github.com/rscada/libmbus/blob/master/debian/libmbus-dev.install#L2

The build script we use used to work.  Is there something we missed when preparing the sources for building as a Debian package?

lategoodbye, 2020-07-15:
Sorry, for the inconvenience. During the conversion of the build system the target for the static library get dropped.

Unfortunately i'm not that cmake expert and partly regret the decision. Nevertheless i started my work, but it's still not usable. Maybe you have an idea what's the problem.

fredrik-sk, 2020-07-15:
We had an issue where the .so file was named liblibmbus.so, I would assume that we removed one "lib" from the name.
So it is possible that the file that previously was called libmbus.a now instead is named mbus.a.

I don't know CMake and do not intend to learn it unless I really have to. So I am of little help.

I tried to build a .deb file, but my "employer" wanted the example programs as well, I was not able to solve that without starting to patch the build files, so I went with a normal .tar.gz where I could do things between the build/installation and the final packaging in an archive file,

lategoodbye, 2020-07-15:
The lib prefix issue has been fixed recently. But in case we don't have a solution at the end of the week, i think about reverting.

lategoodbye, 2020-07-15:
@sjlongland The debian directory is now deprecated. Please take a look at #175 and modify the build.sh to build the debian packages with cmake. I will add the binaries and doc later, after you gave me the okay this is going into right direction.

fredrik-sk, 2020-07-15:
I don't think it because of the lib prefix.

> 
> @sjlongland The debian directory is now deprecated.

Are you certain. I think the the CMake/cpack could use some files from that directory. (But that is only my belife/assumtion.
I don't know cmake, nor deb packages nor cpack

fredrik-sk, 2020-07-15:
I did build the Dockerfile.deb image. It seems that there is a static library being built, and it is presnt in the in the resulting -dev.deb file.

```
# ls -l /tmp/build/lib/libmbus.a
-rw-r--r-- 1 root root 325474 Jul 15 19:56 /tmp/build/lib/libmbus.a

# dpkg -c libmbus_0.9.0-dev.deb
...
-rw-r--r-- root/root    325474 2020-07-15 19:56 ./usr/lib/libmbus.a
...
```

lategoodbye, 2020-07-15:
@fredrik-sk Do you mean before or after my changes?

fredrik-sk, 2020-07-15:
@lategoodbye : After #175

sjlongland, 2020-07-15:
That works, although the build time was unexpectedly short.  A bigger issue is the file names don't conform to Debian packaging conventions.

Nominally these would be named `libmbus_0.9.0_${ARCH}.deb` and `libmbus-dev_0.9.0_${ARCH}.deb` (where `${ARCH}` is the target architecture).  Instead, they're named `libmbus_0.9.0-lib.deb` and `libmbus_0.9.0-dev.deb`.  There's also no source Debian package produced via this method which makes re-building on other architectures easier.

sjlongland, 2020-07-15:
Okay, was able to hack around the package naming:

```
        arch=$( dpkg --print-architecture )
        for f in *.deb; do
            pkgbasename=${f%_*}
            suffix=${f#${pkgbasename}_}
            version=${suffix%-*}
            case "$f" in
                *-lib.deb)
                    mv -v $f ${pkgbasename}_${version}_${arch}.deb
                    ;;
                *-dev.deb)
                    mv -v $f ${pkgbasename}-dev_${version}_${arch}.deb
                    ;;
                *)
                    mv -v $f ${f/.deb/_${arch}.deb}
                    ;;
            esac
        done
```

Until we start getting packages with files named `*_amd64_amd64.deb`, this will get us by I think. :-)

lategoodbye, 2020-07-16:
All those those debuild/debmake/cmake crap, drives me crazy :(

sjlongland, 2020-07-19:
It can be a pain, but try asking your typical end user to do a `./configure && make && make install`.

They just want to press a button and "magic happens", and it doesn't seem to matter whether it's a smartphone user pressing the `install` button on their phone's application repository, or whether its an engineer that wants to integrate M-Bus meters.

lategoodbye, 2020-07-19:
@sjlongland Can you please retest with build-deb.sh?

sjlongland, 2020-07-19:
```
root@bc58b76794df:/tmp/work# ./build-deb.sh 
Running autotools...
configure.ac:19: warning: LT_INIT was called before AM_PROG_AR
aclocal.m4:10027: AM_PROG_AR is expanded from...
configure.ac:19: the top level
…
libtoolize: and rerunning libtoolize and aclocal.
configure.ac:19: warning: LT_INIT was called before AM_PROG_AR
aclocal.m4:10027: AM_PROG_AR is expanded from...
configure.ac:19: the top level
configure.ac:19: warning: LT_INIT was called before AM_PROG_AR
aclocal.m4:10027: AM_PROG_AR is expanded from...
configure.ac:19: the top level
 dpkg-buildpackage -rfakeroot -us -uc -i -b
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: info: source package libmbus
dpkg-buildpackage: info: source version 0.9.0
dpkg-buildpackage: info: source distribution xenial
dpkg-buildpackage: info: source changed by Stefan Wahren <stefan.wahren@i2se.com>
 dpkg-source -i --before-build work
dpkg-buildpackage: info: host architecture amd64
 fakeroot debian/rules clean
dh --with autoreconf clean
dh: Compatibility levels before 9 are deprecated (level 7 in use)
…
configure: creating ./config.status
config.status: creating Makefile
config.status: creating mbus/Makefile
config.status: creating test/Makefile
config.status: creating bin/Makefile
config.status: creating libmbus.pc
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
configure: WARNING: unrecognized options: --disable-maintainer-mode
----------------------------------------------------------
Configuration:

        Source location:        .
        Compile:                gcc
        Compiler flags:         -g -O2
        Linker flags:            -version-info 0:9:0
        Host system type:       x86_64-pc-linux-gnu
        Install path:           /usr

        See config.h for further configuration.
----------------------------------------------------------
   dh_auto_build
dh_auto_build: Compatibility levels before 9 are deprecated (level 7 in use)
        make -j1
make[1]: Entering directory '/tmp/work'
make  all-recursive
make[2]: Entering directory '/tmp/work'
Making all in mbus
make[3]: Entering directory '/tmp/work/mbus'
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -I.. -I..   -g -O2 -c -o mbus.lo mbus.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus.c  -fPIC -DPIC -o .libs/mbus.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus.c -o mbus.o >/dev/null 2>&1
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -I.. -I..   -g -O2 -c -o mbus-protocol.lo mbus-protocol.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-protocol.c  -fPIC -DPIC -o .libs/mbus-protocol.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-protocol.c -o mbus-protocol.o >/dev/null 2>&1
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -I.. -I..   -g -O2 -c -o mbus-tcp.lo mbus-tcp.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-tcp.c  -fPIC -DPIC -o .libs/mbus-tcp.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-tcp.c -o mbus-tcp.o >/dev/null 2>&1
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -I.. -I..   -g -O2 -c -o mbus-serial.lo mbus-serial.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-serial.c  -fPIC -DPIC -o .libs/mbus-serial.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-serial.c -o mbus-serial.o >/dev/null 2>&1
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -I.. -I..   -g -O2 -c -o mbus-protocol-aux.lo mbus-protocol-aux.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I.. -I.. -g -O2 -c mbus-protocol-aux.c  -fPIC -DPIC -o .libs/mbus-protocol-aux.o
[... 186 lines cut ...]
dh_installdeb: Compatibility levels before 9 are deprecated (level 7 in use)
   dh_gencontrol
   dh_md5sums
   dh_builddeb
dpkg-deb: building package 'libmbus-dev' in '../libmbus-dev_0.9.0_amd64.deb'.
dpkg-deb: building package 'libmbus1-dbg' in '../libmbus1-dbg_0.9.0_amd64.deb'.
dpkg-deb: building package 'libmbus1' in '../libmbus1_0.9.0_amd64.deb'.
 dpkg-genbuildinfo --build=binary
 dpkg-genchanges --build=binary >../libmbus_0.9.0_amd64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source -i --after-build work
dpkg-buildpackage: info: binary-only upload (no source included)
Now running lintian...
warning: the authors of lintian do not recommend running it with root privileges!
E: libmbus changes: bad-distribution-in-changes-file xenial
W: libmbus1: package-name-doesnt-match-sonames libmbus0
W: libmbus1: extra-license-file usr/share/doc/libmbus-0.9.0/COPYING
W: libmbus1: manpage-has-bad-whatis-entry usr/share/man/man1/libmbus.1.gz
W: libmbus1: binary-without-manpage usr/bin/mbus-serial-set-address
W: libmbus1: binary-without-manpage usr/bin/mbus-tcp-application-reset
Finished running lintian.
root@bc58b76794df:/tmp/work# 
```

Looking good!  That was on Debian Stretch (v9) via Docker.

sjlongland, 2020-07-19:
Have a look at https://github.com/rscada/libmbus/pull/178 and see what you think.  This adds in source packages, and also allows for signing of packages, should a user require it.