rscada/libmbus #181: Where is the Unit value deriving from?

wmmihaa, 2020-09-14:
We're trying to make sense of the `Unit` field of the data record, which seem to lack consistency. For instance:
* "Volume (m m^3)" 
* "Volume (my m^3)"
* "Volume (1e-1  m^3)"
* "External temperature (1e-2  deg C)"
* "Energy (100 Wh)"
* "C"
* "Time Point (time &amp; date)"

I understand that reverse-engineering the field might never get fully accurate, but if feels like it's just a free text field. Having a look at the [mbus-protocol-aux.c](https://github.com/rscada/libmbus/blob/027f6fb6899b902bdd7d0b3230ecccc24f6bc6c3/mbus/mbus-protocol-aux.c), it looks like pretty structured, but perhaps the Unit value is derived from somewhere else?

In an effort to understand if we would be able to parse the Unit element, I created a function that tries reverse-engineer the text, and then wrote another program that went through all test files in the libmbus repo and extracted every Unit element. Lastly I ran my function on every one of those Units to create the table at the bottom of this issue.

### The report shows three columns:
**Text**       - The Unit value as part of the JSON telegram
**Scale**     - The identified scale value in decimal
**Unit**       - The identified unit value

### The reverse-engineering function works as follows
1.	Ignore Unit fields with no parentheses
2.	If the Unit field has parentheses…
a.	…and the parentheses value can be split on double spaces, then assume the first value is the scale and the second is the unit
b.	…or the parentheses value cannot be split on double spaces, then assume the value is the unit

| Unit field (unparsed)       | Predicted Scale | Predicted Unit |
 | -------------- |-------------|-------------|
|Energy (kWh)|UNKNOWN|kWh|
|Flow temperature (deg C)|UNKNOWN|deg C|
|Return temperature (deg C)|UNKNOWN|deg C|
|Volume flow (m m^3/h)|UNKNOWN|m m^3/h|
|Power (W)|UNKNOWN|W|
|Time Point (time &amp; date)|UNKNOWN|time & date|
|C|UNKNOWN|UNKNOWN|
|c|UNKNOWN|UNKNOWN|
|Fabrication number|UNKNOWN|UNKNOWN|
|Volume (m m^3)|UNKNOWN|m m^3|
|Time Point (date)|UNKNOWN|date|
|Operating time (days)|UNKNOWN|days|
|Firmware version|UNKNOWN|UNKNOWN|
|Software version|UNKNOWN|UNKNOWN|
|On time (days)|UNKNOWN|days|
|Error flags|UNKNOWN|UNKNOWN|
|Volume (my m^3)|UNKNOWN|my m^3|
|Volume (1e-1  m^3)|0.1|m^3|
|Temperature Difference (1e-2  deg C)|0.01|deg C|
|m^3|UNKNOWN|UNKNOWN|
|-|UNKNOWN|UNKNOWN|
|m^3/h|UNKNOWN|UNKNOWN|
|s|UNKNOWN|UNKNOWN|
|°C|UNKNOWN|UNKNOWN|
|W|UNKNOWN|UNKNOWN|
|Wh|UNKNOWN|UNKNOWN|
|K|UNKNOWN|UNKNOWN|
|Energy (Wh)|UNKNOWN|Wh|
|1e-1  V|UNKNOWN|UNKNOWN|
|V|UNKNOWN|UNKNOWN|
|Reset counter|UNKNOWN|UNKNOWN|
|Manufacturer specific|UNKNOWN|UNKNOWN|
|A|UNKNOWN|UNKNOWN|
|cust. ID|UNKNOWN|UNKNOWN|
|bat. time|UNKNOWN|UNKNOWN|
|Flow temperature (1e-1 deg C)|UNKNOWN|1e-1 deg C|
|Return temperature (1e-1 deg C)|UNKNOWN|1e-1 deg C|
|m A|UNKNOWN|UNKNOWN|
|Temperature Difference (m deg C)|UNKNOWN|m deg C|
|Customer location|UNKNOWN|UNKNOWN|
|Digital input (binary)|UNKNOWN|binary|
|1e-2  %RH|UNKNOWN|UNKNOWN|
|External temperature (1e-2  deg C)|0.01|deg C|
|Special supplier information|UNKNOWN|UNKNOWN|
|Averaging Duration (hours)|UNKNOWN|hours|
|Digital output (binary)|UNKNOWN|binary|
| V|UNKNOWN|UNKNOWN|
|Energy (10 Wh)|UNKNOWN|10 Wh|
|1e-1  A|UNKNOWN|UNKNOWN|
|Power (10 W)|UNKNOWN|10 W|
|Volume (1e-2  m^3)|0.01|m^3|
|Units for H.C.A.|UNKNOWN|UNKNOWN|
|Temperature Difference (1e-1  deg C)|0.1|deg C|
|Power (100 W)|UNKNOWN|100 W|
|On time (hours)|UNKNOWN|hours|
|Volume ( m^3)|UNKNOWN|m^3|
|Operating time (hours)|UNKNOWN|hours|
|Volume flow (1e-2  m^3/h)|0.01|m^3/h|
|Power (1e-1 W)|UNKNOWN|1e-1 W|
|Energy (10 kWh)|UNKNOWN|10 kWh|
|Volume flow ( m^3/h)|UNKNOWN|m^3/h|
|On time (seconds)|UNKNOWN|seconds|
|Operating time (seconds)|UNKNOWN|seconds|
|Power (kW)|UNKNOWN|kW|
|Temperature Difference ( deg C)|UNKNOWN|deg C|
|1e-2  V|UNKNOWN|UNKNOWN|
|dimensionless / no VIF|UNKNOWN|UNKNOWN|
|Energy (mWh)|UNKNOWN|mWh|
|l|UNKNOWN|UNKNOWN|
|reserved but historic|UNKNOWN|UNKNOWN|
|External temperature ( deg C)|UNKNOWN|deg C|
|Energy (100 Wh)|UNKNOWN|100 Wh|
|Volume (1e-4  m^3)|0.0001|m^3|
|Volume flow (1e-4  m^3/h)|0.0001|m^3/h|
|Flow temperature (1e-2 deg C)|UNKNOWN|1e-2 deg C|
|Return temperature (1e-2 deg C)|UNKNOWN|1e-2 deg C|
|Volume flow (1e-1  m^3/h)|0.1|m^3/h|
|Energy (0.1 MWh)|UNKNOWN|0.1 MWh|
|Model / Version|UNKNOWN|UNKNOWN|
|Parameter set identification|UNKNOWN|UNKNOWN|
|Reserved VIF extension|UNKNOWN|UNKNOWN|
|kWh|UNKNOWN|UNKNOWN|
|Unknown (VIF=0x7B)|UNKNOWN|VIF=0x7B|
|Energy (MJ)|UNKNOWN|MJ|
|Unknown (VIF=0x79)|UNKNOWN|VIF=0x79|
|Medium (as in fixed header)|UNKNOWN|as in fixed header|
|Actuality Duration (seconds)|UNKNOWN|seconds|
|Averaging Duration (seconds)|UNKNOWN|seconds|
|Averaging Duration (minutes)|UNKNOWN|minutes|
|Reserved|UNKNOWN|UNKNOWN|
|J|UNKNOWN|UNKNOWN|

wmmihaa, 2020-09-15:
After further investigation it seems I'm using the VIF output from mbus-protocol.c, rather than mbus-protocol-aux.c. Is there a setting or compile flag I can use?

lategoodbye, 2020-09-15:
In order to use mbus-protocol-aux decoding, you can use test/mbus_parse_hex which support the parameter -n. The default output of mbus-protocol is more intended for human. AFAIK the M-Bus protocol wasn't designed for machines.

wmmihaa, 2020-09-15:
We're just using the `mbus_is_secondary_address` method. Is there a compile flag or some other global options that I can set to force using `mbus_vif_unit_normalize`?

PS(2). http://www.rscada.se/ is not accessable....(internal server error)