rscada/libmbus #184: m.bus Adapter - Current Unrecognized VIF extension: 0x3a - Stromzähler B+G Tech SDM 630 V2

Habedere, 2020-10-29:
Hallo,
ich möchte aktuell o.g. Stromzähler "SDM 630" auslesen.
Der Mbus Adapter findet die Zähler und erstellte auch Objekte.
Allerdings kann ich nicht alle Werte auslesen.
Bei einigen steht wie folgt:
Name "Current Unrecognized VIF extension: 0x3a"
Wert "11296 Unrecognized VIF extension: 0x3a"

[image]

Eine Idee woran es liegen könnte?
Google meint evtl. sind mehr Daten auszulesen wenn man über die Sekundäradresse ausliest. Macht das Sinn?
Kann ich hier etwas umstellen?

Apollon77, 2020-10-29:
mbus nodejs lib is based on current master version of the lib (exactly 16.7.2020)

Habedere, 2020-11-30:
No idea?

Habedere, 2021-01-24:
Hat keiner einen Ansatzpunkt für mich?

Habedere, 2021-09-12:
Some news here?