rscada/libmbus #187: Mbus clarification

NJSannuvanda, 2021-05-13:
Q : This is to make sure if the Mbus provide Stack internally ?(Preloaded)

lategoodbye, 2021-05-13:
@NJSannuvanda Sorry, i didn't understand your question. Can you please explain?

NJSannuvanda, 2021-05-13:
Regarding libmubs open source code - 

We are looking for an Mbus stack to be implemented in one of our project.
So need to know whether this Code provide Stack Preloaded or need to keep in microcontroller

And also can i know the Part number of the chip?

lategoodbye, 2021-05-13:
This project wasn't intended for small microcontroller since it requires a bigger OS e.g. Linux for serial communication.

Not sure what chip you mean. This is just a open source software ṕroject, there are no real specific hardware dependencies.