rscada/libmbus #191: Negative values start with F

trsqr, 2021-12-22:
I've got a Techem Ultra S3 (which is basically a rebranded Sharky 775) heat meter. For some reason the negative values show up prefixed with F. Is this expected?

Here for example we should have a negative power of 99 watts:

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (W)</Unit>
        <Value>F0000099</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

This didn't happen previously on my old system, but unfortunately that one crashed, so I can't really go back and see which version of libmbus I was running there.

```
$ mbus-tcp-request-data -d 192.168.99.15 10001 7
[2021-12-22 08:56:00Z] SEND (005): 10 5B 07 62 16
[2021-12-22 08:56:01Z] RECV (110): 68 68 68 68 08 07 72 45 11 06 48 68 50 40 04 8C 00 00 00 0C 06 26 25 10 00 8C 10 06 00 00 00 00 8C 20 13 43 55 64 08 0C 13 56 13 30 36 0C 2B 99 00 00 F0 0B 3B 71 09 00 0A 5A 20 03 0A 5E 21 03 0A 62 00 F0 0A 27 66 29 04 6D 2A 0C B6 2C 4C 06 85 14 09 00 4C 13 82 71 73 28 CC 10 06 00 00 00 00 CC 20 13 11 95 61 05 42 6C 9F 2C EC 16
mbus_frame_print: Dumping M-Bus frame [type 4, 110 bytes]: 68 68 68 68 08 07 72 45 11 06 48 68 50 40 04 8C 00 00 00 0C 06 26 25 10 00 8C 10 06 00 00 00 00 8C 20 13 43 55 64 08 0C 13 56 13 30 36 0C 2B 99 00 00 F0 0B 3B 71 09 00 0A 5A 20 03 0A 5E 21 03 0A 62 00 F0 0A 27 66 29 04 6D 2A 0C B6 2C 4C 06 85 14 09 00 4C 13 82 71 73 28 CC 10 06 00 00 00 00 CC 20 13 11 95 61 05 42 6C 9F 2C EC 16
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>48061145</Id>
        <Manufacturer>TCH</Manufacturer>
        <Version>64</Version>
        <ProductName></ProductName>
        <Medium>Heat: Outlet</Medium>
        <AccessNumber>140</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>102526</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>1</Tariff>
        <Device>0</Device>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>2</Tariff>
        <Device>0</Device>
        <Unit>Volume (m m^3)</Unit>
        <Value>8645543</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>36301356</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Power (W)</Unit>
        <Value>F0000099</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume flow (m m^3/h)</Unit>
        <Value>971</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Flow temperature (1e-1 deg C)</Unit>
        <Value>320</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Return temperature (1e-1 deg C)</Unit>
        <Value>321</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Temperature Difference (1e-1  deg C)</Unit>
        <Value>F000</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Operating time (days)</Unit>
        <Value>2966</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2021-12-22T12:42:00</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>91485</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="12">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>28737182</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="13">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Tariff>1</Tariff>
        <Device>0</Device>
        <Unit>Energy (kWh)</Unit>
        <Value>0</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="14">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Tariff>2</Tariff>
        <Device>0</Device>
        <Unit>Volume (m m^3)</Unit>
        <Value>5619511</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="15">
        <Function>Instantaneous value</Function>
        <StorageNumber>1</StorageNumber>
        <Unit>Time Point (date)</Unit>
        <Value>2020-12-31</Value>
        <Timestamp>2021-12-22T08:56:01Z</Timestamp>
    </DataRecord>

</MBusData>

```

lategoodbye, 2021-12-22:
@trsqr Thanks for your report. Is it okay for you to let us integrate the M-Bus telegram in libmbus test data?

trsqr, 2021-12-22:
Absolutely.

lategoodbye, 2021-12-27:
Today i had some time to look into this. At first the behavior of mbus-tcp-request-data is expected, because most of the examples uses the non-normalized output which is dedicated for humans (like M-Bus in general). Since this commit 2f9fa5ccc8aa032556ab77b7cafcb24670b6bf8d the libmbus is able to handle F as a negative value.

So your next question would be like: why doesn't mbus-tcp-request-data use this feature?
Answer: it's not that simple. Unfortunately all the other hex values A-E are also valid values, so simply enabling this for every BCD value would break error values. So it's very hard for a machine to decided how to decode.

A possible hack could be to handle "Value during error state" as hex value and the other one as integer.

lategoodbye, 2021-12-27:
The hack is available as PR #197