rscada/libmbus #199: Documentation not accesible

Mk-arc, 2022-05-10:
Hi,

i am pretty new to mbus. And the documentation website seems to not respond? Is there a mirror anywhere?
visitng http://www.rscada.se/ just gives an error. 

I got a cheap usb to mbus converter and hooked it up with an Relay PadPulse M2: https://www.relay.de/produkte/m-bus-slaves/padpuls-m2

When I try to run the scan I get following result:

```
./mbus-serial-scan -d /dev/ttyUSB0 
Scanning primary addresses:
0 [2022-05-10 21:25:53Z] SEND (005): 10 40 00 40 16
[2022-05-10 21:25:53Z] RECV (001): A5
Collision at address 0
1 [2022-05-10 21:25:53Z] SEND (005): 10 40 01 41 16
2 [2022-05-10 21:25:53Z] SEND (005): 10 40 02 42 16
3 [2022-05-10 21:25:53Z] SEND (005): 10 40 03 43 16
4 [2022-05-10 21:25:54Z] SEND (005): 10 40 04 44 16
5 [2022-05-10 21:25:54Z] SEND (005): 10 40 05 45 16
6 [2022-05-10 21:25:54Z] SEND (005): 10 40 06 46 16
7 [2022-05-10 21:25:54Z] SEND (005): 10 40 07 47 16
8 [2022-05-10 21:25:55Z] SEND (005): 10 40 08 48 16
```

How to get started with figuring out whats wrong?

closed 2022-05-13 by Mk-arc

lategoodbye, 2022-05-14:
Is it possible there are multiple M-Bus slaves connected with the same primary id? Did you tried scanning for secondary address?

Mk-arc, 2022-05-14:
Hi @lategoodbye yes. It worked when I send data requests to address 254 and also set address worked and then the direct access without broadcast also worked fine. So I closed this issue.

Thanks!