rscada/libmbus #204: Sending different init frames to a slave...

Gubelat, 2022-10-20:


Apollon77, 2022-10-21:
What should this issue tell anyone? @Gubelat

Gubelat, 2022-10-21:
Sorry, i was not able to delete this headline. and it was late Yesterday.
The Problem is that my Heatmeter is running on batteries. So it restrict the amount of readouts in standard telegram mode.
If i run the serial meter readout tool, i am only can poll data every 69 minutes for standard telegrams. If i run it for short telegrams i can run the readout every 5 minutes without restrictions.

So i need to send a special init Frame to the meter to get this short telegrams.

The sequence looks like this:

```
Switching to short telegram mode: 1 response telegram:
68 04 04 68 73 PB 50 50 CS   or
68 04 04 68 73 PB 53 50 CS
"50” means sending instantaneous values (for controls)

PB = Primary Bus Adress
CS = Checksum

```
The question is now how to edit the source code for sending this init Frame to the Meter?
There is no command line option to allow sending other init frames.

German Original Specification:

[Mbus telegramm spec sensonic 3_08.09.pdf](https://github.com/rscada/libmbus/files/9838635/Mbus.telegramm.spec.sensonic.3_08.09.pdf)

Englisch Translation (Gooogle)
[Mbus telegramm spec sensonic 3_08.09.de.en.pdf](https://github.com/rscada/libmbus/files/9838725/Mbus.telegramm.spec.sensonic.3_08.09.de.en.pdf)