rscada/libmbus #207: Received invalid M-Bus response frame

ptorrent, 2023-04-03:
Hello there !

I'm not able to get device info:

```
mbus-serial-request-data-multi-reply -d -b 2400 /dev/ttyS1 2

init_slaves: debug: sending init frame #1
[2023-04-03 16:54:25Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2023-04-03 16:54:25Z] SEND (005): 10 40 FD 3D 16
[2023-04-03 16:54:25Z] SEND (005): 10 7B 02 7D 16
mbus_sendrecv_request: No M-Bus response frame received.
[2023-04-03 16:54:26Z] SEND (005): 10 7B 02 7D 16
mbus_serial_recv_frame: Timeout
[2023-04-03 16:54:27Z] RECV (168): 68 BE BE 68 08 02 72 49 25 91 05 B4 05 C4 04 52 00 00 00 04 86 3B FC 05 00 00 04 86 3C 33 00 00 00 00 05 2B 00 00 00 80 05 3B 00 00 00 00 05 5B 18 DF B0 41 05 5F C8 B6 BD 41 05 63 00 7B CD BF 05 83 33 73 5E 84 3F 05 9B 2C 48 DC 85 3F 04 22 98 10 00 00 34 22 01 00 00 00 B4 40 22 01 00 00 00 04 6D 32 33 E3 24 05 93 28 00 00 20 41 85 40 93 28 00 00 80 3F 85 80 40 93 28 00 00 80 3F 0C 78 49 25 91 05 0D FD 11 10 65 72 69 61 6C 6F 53 20 72 75 65 74 70 6D 6F 43 0B FD 0E 01 01 02 0C FD 0D 13 00 12 02 FE 16
mbus_sendrecv_request: Received invalid M-Bus response frame.
[2023-04-03 16:54:27Z] SEND (005): 10 7B 02 7D 16
mbus_sendrecv_request: No M-Bus response frame received.
[2023-04-03 16:54:27Z] SEND (005): 10 7B 02 7D 16
mbus_serial_recv_frame: Timeout
[2023-04-03 16:54:28Z] RECV (169): 68 BE BE 68 08 02 72 49 25 91 05 B4 05 C4 04 53 00 00 00 04 86 3B FC 05 00 00 04 86 3C 33 00 00 00 00 00 05 2B 00 00 00 80 05 3B 00 00 00 00 05 5B 58 D5 B0 41 05 5F A8 B2 BD 41 05 63 00 D5 CD BF 05 83 33 63 5E 84 3F 05 9B 2C 4B DC 85 3F 04 22 98 10 00 00 34 22 01 00 00 00 B4 40 22 01 00 00 00 04 6D 32 33 E3 24 05 93 28 00 00 20 41 85 40 93 28 00 00 80 3F 85 80 40 93 28 00 00 80 3F 0C 78 49 25 91 05 0D FD 11 10 65 72 69 61 6C 6F 53 20 72 75 65 74 70 6D 6F 43 0B FD 0E 01 01 02 0C FD 0D 13 00 12 02 5E 16
mbus_sendrecv_request: Received invalid M-Bus response frame.
Failed to send/receive M-Bus request.

```

Some of them is working, other not.. all on the same line.

Do you have an idea ?

Thanks for your support

ptorrent, 2023-04-04:
Solving by increasing serial timeout...

[image]

slsimic, 2024-09-09:
It would be nice if timeout could be issued as command line parameter.
I can see with -d option when I do mbus-tcp-scan that SND_NKE requests are sent at 4s interval so I presume 4 seconds is default timeout for TCP requests.
But sometimes, especially when slaves aren't on the local network or accessed via VPN thru GPRS, timeouts must be higher than 4 seconds.
Sometimes, much higher, depending on network quality, mobile signal or mbus slave speed. 
300bps answer (which is fallout standard) certainly cannot go thru with 4s timeout with REQ_UD2 requests.