rscada/libmbus #212: libmbus problem in python-script

kptkip, 2023-12-16:
I have problems within a python script that uses the libmbus library:

the script looks like this:
```
#!/usr/bin/env python3

import sys
import json
import paho.mqtt.client as mqtt

import xml.etree.ElementTree as ET

from mbus.MBus import MBus

usbdevice = sys.argv[1] # /dev/ttyAMA0 
address = sys.argv[2]   # 0 

mbus = MBus(device=usbdevice)

mbus.connect()
mbus.send_request_frame(int(address))

reply = None

try:
        reply = mbus.recv_frame()
except:
        sys.exit(1)

reply_data = mbus.frame_data_parse(reply)

xml = mbus.frame_data_xml(reply_data).strip()

#print(xml)

mbus.frame_data_free(reply_data)
mbus.disconnect()

# MQTT it
#
tree = ET.ElementTree(ET.fromstring(xml))

Meter        = int(tree.findall(".//DataRecord[@id='1']/Value")[0].text)
ThermalPower = int(tree.findall(".//DataRecord[@id='3']/Value")[0].text)
Flow         = int(tree.findall(".//DataRecord[@id='5']/Value")[0].text)
Temp_VL      = int(tree.findall(".//DataRecord[@id='7']/Value")[0].text)
Temp_RL      = int(tree.findall(".//DataRecord[@id='8']/Value")[0].text)
Temp_Diff    = int(tree.findall(".//DataRecord[@id='9']/Value")[0].text)

client = mqtt.Client()
client.connect("192.168.178.89", 1883, 60)

client.publish("panasonic_heat_pump/Mbus/Meter",Meter)
client.publish("panasonic_heat_pump/Mbus/ThermalPower",ThermalPower)
client.publish("panasonic_heat_pump/Mbus/Flow",Flow)
client.publish("panasonic_heat_pump/Mbus/Temp_VL",Temp_VL)
client.publish("panasonic_heat_pump/Mbus/Temp_RL",Temp_RL)
client.publish("panasonic_heat_pump/Mbus/Temp_Diff",Temp_Diff)

client.disconnect()
```

The system reacts like this:

```
Traceback (most recent call last):
  File "/home/pi/ReadHRIDataUnitMBus.py", line 18, in <module>
    usbdevice = sys.argv[1] # /dev/ttyAMA0 
                ~~~~~~~~^^^
IndexError: list index out of range
pi@MBusPI:~ $ /usr/bin/python3 /home/pi/ReadHRIDataUnitMBus.py /dev/serial0 0
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/mbus-0.0.1-py3.11.egg/mbus/MBus.py", line 66, in __init__
  File "/usr/local/lib/python3.11/dist-packages/mbus-0.0.1-py3.11.egg/mbus/MBusLowLevel.py", line 199, in __init__
  File "/usr/lib/python3.11/ctypes/__init__.py", line 389, in __getattr__
    func = self.__getitem__(name)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 394, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: /usr/local/lib/libmbus.so.0: undefined symbol: mbus_get_current_version

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/ReadHRIDataUnitMBus.py", line 21, in <module>
    mbus = MBus(device="-b 2400 /dev/serial0")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/mbus-0.0.1-py3.11.egg/mbus/MBus.py", line 68, in __init__
OSError: libmbus not found
Exception ignored in: <function MBus.__del__ at 0x75ed24d8>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/mbus-0.0.1-py3.11.egg/mbus/MBus.py", line 97, in __del__
AttributeError: 'MBus' object has no attribute 'handle'
```
Anyone a clue to tackle this?
Thanks

Apollon77, 2023-12-18:
All the errors are in pytho parts and this library here is not the python one. issue created in right repo?

kptkip, 2023-12-18:
This could be that it is the wrong repo ;-)
I assumed, that this is the right place because of this message:
`OSError: libmbus not found`
And to be honest, I had no clue, where to search for.

But at the end I found a solution with a shell-script that fetches the mbus data via `mbus-serial-request-data` and brings it via mosquitto to my IoBroker system.

Thanks for the response.

closed 2023-12-18 by kptkip