rscada/libmbus #216: Receiving FF byte over and over again

LordGregorious, 2024-07-25:
Hi. I hope someone is still out there to help. 

I have been using libmbus on raspberry pis to communicate with a [Relay PW250](https://www.relay.de/produkte/m-bus-master/pegelwandler-pw-250) to read water meters. I have several sites with this set up and it's always worked fine. 

Until now. When I run mbus-serial-request-data at a baudrate of 2400 I jus receive a 00 byte (not always), followed by the FF byte over and over again. It's a similar story with the mbus-serial-scan. 

However when I connect my windows laptop to the PW250 and run MBSheet (at 2400 baudrate) it works perfectly fine. 
I've tried with two different raspberry pis ( raspberry pi 3 and raspberry pi 5), tried different usb ports, tried with a different cable. The problem persists.

I've desperately searched and tried things but nothing has worked. Any help would be immensely appreciated. I can post logs if you need.

cybermailer84, 2024-09-23:
which raspbian do you use. I have the same problem since my last updates. I will try an older raspbian version and have to check. I dont know exactly if something was changed and that causes the problem, but i think so.

LordGregorious, 2024-10-07:
The raspberry pi is running Debian GNU/Linux 12 (bookworm). 
I found the issue was that one of the modules was water-damaged (or something) and was constantly sending out rubbish data, thus clogging the bus and all communication. 

I recommend disconnecting your network of modules from the repeater and just testing a single module. 
If it works fine then likely one of the modules is faulty and you'll have to manually go through them all to find it. 

For me, it was set up floor by floor in an apartment complex, so I could test each floor individually until I found the one containing the faulty module.